DreamSourceLab/DSView #489: UART decoder : can't filter bits from export file

plankton321, 2022-05-04:
UART decoder works fine as it does in Pulse View with one exception. No matter what I deselect I cannot remove the bits (stop, start, etc) from the exported file.  All I want/need are the decoded values in hex. That's it. I don't want break, party, start/stop however the deselect in setup doesn't work and I get those values anyway.  Can someone please fix or show me how to deselect/configure so only export the decoded values.  Thanks  Scott

plankton321, 2022-05-04:
[image]

[image]

dreamsource-tai, 2022-05-05:
@plankton321 
Disable the ‘Display start/stop’ option in the setting dialog.