DreamSourceLab/DSView #490: DSView cannot run 22.04 with segmentation fault

priv, 2022-05-04:
The crash stack does not tell anything useful

```
(gdb) run
Starting program: /usr/local/bin/DSView 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1bc4640 (LWP 1880104)]
[New Thread 0x7fffebfbc640 (LWP 1880160)]
[New Thread 0x7fffeb7bb640 (LWP 1880165)]
[New Thread 0x7fffeae66640 (LWP 1880199)]
[New Thread 0x7fffea665640 (LWP 1880207)]
srd: Failed to load decoder ir_sirc: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder sle44xx: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder ir_sirc: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder sle44xx: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

[New Thread 0x7fffe93db640 (LWP 1880320)]
[New Thread 0x7fffe8bda640 (LWP 1880322)]
[New Thread 0x7fffdbfff640 (LWP 1880324)]
[New Thread 0x7fffdb7fe640 (LWP 1880325)]

Thread 1 "DSView" received signal SIGSEGV, Segmentation fault.
___pthread_mutex_lock (mutex=0xa0) at ./nptl/pthread_mutex_lock.c:80
80      ./nptl/pthread_mutex_lock.c: No such file or directory.
```

dreamsource-tai, 2022-05-05:
@benner 
@priv 
You can pull the source code,and switch to "dev-tai" branch,recompile the project.We have fixed the python package of common.

priv, 2022-05-05:
I can confirmed that 1.2.0-rc6 does not crash on Ubuntu 22.04.

I don't have dscope or dslogic device on hand today so I'll test another day.

benner, 2022-05-05:
I can confirm it too. Build specified branch and it works. Thank you

priv, 2022-05-05:
Close this issue since already fixed in rc.

closed 2022-05-05 by priv