DreamSourceLab/DSView #491: SPI repeats sampling for a period of time and then blinks back

kktornado, 2022-05-05:
Set the repetition interval to 1.SPI repeats sampling for a period of time and then flashes back

dreamsource-tai, 2022-05-05:
@kktornado 
你的系统和软件版本是？

kktornado, 2022-05-06:
版本	Windows 10 专业版
版本号	21H2
安装日期	‎2021/‎3/‎3
操作系统内部版本	19044.1645
序列号	PF2ML6X5
体验	Windows Feature Experience Pack 120.2212.4170.0

dreamsource-tai, 2022-05-06:
@kktornado 
能再现这个问题吗，可以的话我们远程看一下你说的问题。请关注公众号，联系客服，并说明来意。

boris-chen-github, 2022-05-19:
Hi dreamsource-tai,

這問題我也有遇過, 您可以試試看錄製訊號(ex: MDC/MDIO)超過半小時以上, 此時DSview會非常卡頓。
另外我想詢問, 何時會有新版本的UI更新呢? 官網的UI已超過兩年未更新了。但是github這邊直到七個月前都還有變動。

Thanks!

dreamsource-tai, 2022-05-27:
@boris-chen-github 
目前我们在做内部测试，近期会发布一个新版本，到时候会有新的体验，请留意哦。谢谢！
至于卡的问题，这个要看硬件配置，特殊情况下，内存要够大才行。

dreamsource-tai, 2022-05-27:
@boris-chen-github 
我们整个项目都是开源的，协议也是开源的，任何人都可以修改和添加。你要是有这个需求，可以参考i2c，然后自己开发出i3c。可以的话，到时候希望你也贡献一下你的代码哦。
协议教程的话，可以关注微信公从号“梦源科技”，上面有教程。下面的链接也能打开：
https://wap.qidian.qq.com/ol/rest/view/3009108502_11402_1_1650351587