DreamSourceLab/DSView #492: Apple Silicon机器编译libsigrok4DSL出错

huming2207, 2022-05-11:
您好，

本人有一台MacBook Pro M1，目前使用官网最新的x86版本的DSView 1.1.2，能正常使用，但比较费电，所以想试着编译一个arm64版，但目前出现以下问题：

1. 在编译libsigrok4DSL时报错找不到glib，已尝试`brew install glib`安装但依然报错：

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in hardware
Making all in demo
  CC       libsigrok4DSL_hw_demo_la-demo.lo
clang: warning: -ludev: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--rpath=/usr/local/lib: 'linker' input unused [-Wunused-command-line-argument]
In file included from demo.c:23:
./demo.h:24:10: fatal error: 'glib.h' file not found
#include <glib.h>
         ^~~~~~~~
1 error generated.
make[3]: *** [libsigrok4DSL_hw_demo_la-demo.lo] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
```

2. 编译libsigrokdecode4DSL报错如下：

```
  CC       version.lo
type_decoder.c:63:6: warning: unused variable 'ijmp' [-Wunused-variable]
        int ijmp = 0;
            ^
type_decoder.c:59:12: warning: unused variable 'py_item' [-Wunused-variable]
        PyObject *py_item, *py_bytes;
                  ^
srd.c:255:2: warning: 'PyEval_InitThreads' is deprecated [-Wdeprecated-declarations]
        PyEval_InitThreads();
        ^
/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9/ceval.h:130:1: note: 'PyEval_InitThreads' has been explicitly marked deprecated here
Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
^
/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9/pyport.h:508:54: note: expanded from macro 'Py_DEPRECATED'
#define Py_DEPRECATED(VERSION_UNUSED) __attribute__((__deprecated__))
                                                     ^
decoder.c:1069:6: warning: variable 'ldst' set but not used [-Wunused-but-set-variable]
        int ldst = 0;
            ^
instance.c:1054:10: warning: unused variable 'error_buffer' [-Wunused-variable]
        char ** error_buffer = NULL;
                ^
1 warning generated.
2 warnings generated.
1 warning generated.
1 warning generated.
  CCLD     libsigrokdecode4DSL.la
Undefined symbols for architecture arm64:
  "_PyArg_ParseTuple", referenced from:
      _Decoder_put in type_decoder.o
      _Decoder_wait in type_decoder.o
      _Decoder_has_channel in type_decoder.o
      _Decoder_printlog in type_decoder.o
  "_PyArg_ParseTupleAndKeywords", referenced from:
      _Decoder_register in type_decoder.o
  "_PyBytes_AsString", referenced from:
      _srd_init in srd.o
      _di_thread in instance.o
      _py_str_as_str in util.o
      _py_strseq_to_char in util.o
      _py_obj_to_variant in util.o
      _srd_exception_catch in exception.o
      _Decoder_put in type_decoder.o
      ...
[... 257 lines cut ...]
  "__Py_Dealloc", referenced from:
      _srd_init in srd.o
      _srd_decoder_searchpath_add in srd.o
      _srd_inst_send_meta in session.o
      _srd_session_end in session.o
      _srd_decoder_apiver in decoder.o
      _srd_decoder_load in decoder.o
      _check_method in decoder.o
      ...
  "__Py_FalseStruct", referenced from:
      _Decoder_has_channel in type_decoder.o
  "__Py_NoneStruct", referenced from:
      _srd_decoder_doc_get in decoder.o
      _Decoder_put in type_decoder.o
      _Decoder_wait in type_decoder.o
      _Decoder_printlog in type_decoder.o
  "__Py_TrueStruct", referenced from:
      _Decoder_has_channel in type_decoder.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libsigrokdecode4DSL.la] Error 1
make: *** [all] Error 2
```

想请教一下如何处理呢？谢谢！

dreamsource-tai, 2022-05-11:
@huming2207 
mac下的包环境多样，使有的用户在编译过程中包的检测会失败。我们后面会把这个修改一下。
你如果急的话，可以关注公众号”梦源科技“，联系客服并说明问题，然后会给你发一个QT的工程文件，你通过qtcreator来编译，qtcreator没有找不到包这方面的问题。

huming2207, 2022-05-11:
@dreamsource-tai 感谢回复，这个不急，目前x86版也能凑合用，只是性能差而且比较费电，先等您那边修复了。同时我也建议下一个版本也麻烦提供一下编译好的arm64版DSView，谢谢。

dreamsource-tai, 2022-05-18:
@huming2207 
你好，我们修复了在mac os下的编译脚本，请你拉取最新的代码，切换到分支 "dev-tai"，然后编译一下，我们对编译脚本进行了简化。

huming2207, 2022-05-18:
@dreamsource-tai 刚看到了，

似乎您这个glib还有点问题：

[image]

huming2207, 2022-05-18:
@dreamsource-tai 提示如下：

···
ld: warning: directory not found for option '-L/Users/hu/Downloads/DSView/glib-2.0'
ld: warning: directory not found for option '-L/Users/hu/Downloads/DSView/intl'
ld: warning: -L path '/usr/local/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib' is not a directory
ld: warning: -L path '/opt/homebrew/lib/libfftw3.dylib' is not a directory
ld: warning: -L path '/opt/homebrew/lib/libusb-1.0.dylib' is not a directory
ld: warning: -L path '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/usr/lib/libz.tbd' is not a directory
ld: warning: directory not found for option '-L/Users/hu/Downloads/DSView/zip'
ld: library not found for -lglib-2.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [build/DSView] Error 1
make[1]: *** [CMakeFiles/DSView.dir/all] Error 2
make: *** [all] Error 2
···

dreamsource-tai, 2022-05-18:
@huming2207 
我们这边确实是能成功编译了的。你以源码方式安装glib2.0试试看。可以的话我们加微信啊，沟通方便些。通过”梦源科技“公众号，问客服要微信号。

huming2207, 2022-05-18:
@dreamsource-tai 好的明白，我明天弄一下，到时候要还是不行的话就联系您，谢谢

huming2207, 2022-05-19:
@dreamsource-tai 您好，我这边已经在公众号上给客服留言了，烦请看一下，谢谢。

另外这里是我编译的log: https://gist.github.com/huming2207/9a11b8e6d0121bcd5ed260eaee6f5c71

似乎问题出在您那个cmake依赖检测的部分，我手动把`GLIB_LIBRARIES`之类的变量改成绝对路径就可以了。最终成品如下：

[image]

但目前我是直接make install的，不太清楚如何打包此类macOS应用成app bundle，但也不是不能用，就凑合用着吧。

dreamsource-tai, 2022-05-19:
@huming2207 
make编译的目录结构跟linux下一样，所以就没做成打包了，谢谢反馈。

huming2207, 2022-05-22:
@dreamsource-tai 您好，我发现那个python的问题是因为cmake识别到了我装在Rosetta模式下的homebrew的python 3.9，也就是说那是x86_64的python。也就是说之前我机子上是有两个homebrew的，一个是arm64原生homebrew，在`/opt`下面，还有个x86_64的homebrew，在`/usr/local`下面。

鉴于目前homebrew的arm64源已经非常完善，我刚就把x86的homebrew给删掉了。但删掉以后再pull您master （e3efc5ff769f0041e852be2ffd655e6a1586fb87）或者dev-tai分支的代码，会提示找不到python：

```
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2")
-- Checking for one of the modules 'glib-2.0'
-- glib-2.0:
-- 	- includes:/opt/homebrew/Cellar/glib/2.72.1/include/glib-2.0/opt/homebrew/Cellar/glib/2.72.1/lib/glib-2.0/include/opt/homebrew/opt/gettext/include/opt/homebrew/Cellar/pcre/8.45/include
-- 	- libraries:/opt/homebrew/Cellar/glib/2.72.1/lib/libglib-2.0.*
CMake Error at /opt/homebrew/Cellar/cmake/3.23.1_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.23.1_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.23.1_1/share/cmake/Modules/FindPythonLibs.cmake:310 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:91 (find_package)
```

dreamsource-tai, 2022-05-27:
@huming2207 
不好意思，才看到你的消息。是的，在m1下会分x86和arm的库，你的终端不要以Rosetta运行，然后重新安装brew，这个时候的brew就不是x86的。你想编译arm的话，应该卸载掉x86的brew,且把x86的所有库先卸载掉，然后再试安装各种库，这个时候再来编译，没问题的话，它应该是原生的arm程序。现在master分支是代码是最新的，dev-tai分支现在没有更新了。

huming2207, 2022-05-27:
@dreamsource-tai 目前最新master还是提示找不到python，如下：

```
CMake Deprecation Warning at CMakeLists.txt:22 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2")
-- Checking for one of the modules 'glib-2.0'
-- glib-2.0:
-- 	- includes:/opt/homebrew/Cellar/glib/2.72.1/include/glib-2.0/opt/homebrew/Cellar/glib/2.72.1/lib/glib-2.0/include/opt/homebrew/opt/gettext/include/opt/homebrew/Cellar/pcre/8.45/include
-- 	- libraries:/opt/homebrew/Cellar/glib/2.72.1/lib/libglib-2.0.*
CMake Error at /opt/homebrew/Cellar/cmake/3.23.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.23.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.23.2/share/cmake/Modules/FindPythonLibs.cmake:310 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:91 (find_package)

```

似乎这是macOS aarch64 homebrew下独有的问题，见此参考：https://github.com/google-research/football/issues/233

dreamsource-tai, 2022-05-27:
@huming2207 
你先卸载这个python，然后重装试试看。再来看pkg-config目录下有没有PythonLibs.pc文件。实在不行，可以手工修改cmakelists.txt文件，关于python那一块的逻辑。
pkg-config应试在某个lib目录下。

huming2207, 2022-05-27:
> 你先卸载这个python，然后重装试试看。再来看pkg-config目录下有没有PythonLibs.pc文件。实在不行，可以手工修改cmakelists.txt文件，关于python那一块的逻辑。

这个我前几天就已经删掉了，整个x86版的brew都干掉了，所以不应该存在这个问题

huming2207, 2022-06-29:
@dreamsource-tai 您好，刚试了下，现在已经能正常编译了！非常感谢！

closed 2022-06-29 by huming2207