DreamSourceLab/DSView #493: Repetitive interval less than 1 second

imyaFamiliev, 2022-05-18:
Hello!
I need to measure the phase between signal correlated to  PPS (pulse per second)  from GPS. Front of PPS using as a trigger source. Capture mode is Repetitive.
Using custom decoder i write this data to file continuously, BUT with repetitive interval 1 second dsview can only save events with 2 second between them. So i think, if i can set interval about 0.5s - 0.9s (but not lower than session length of course) I able to capture all pps evens. Can you change this settings?
Thanks!

dreamsource-tai, 2022-05-20:
@imyaFamiliev 
        What is your operating system? If it is Linux, please pull the latest code and compile it. If it's windows, please wait until we release the next version.On the toolbar, click the mode button, select the repetition mode, and enter the appropriate time value. 0.1 equals 100 milliseconds.

imyaFamiliev, 2022-05-20:
Thanks for such a fast response! I use windows 10, so I`ll wait for new win release.