DreamSourceLab/DSView #494: Wrong I2C decoding. Some bytes reassembled from 7 bits instead of 8.

BrainsLab, 2022-05-23:
Hi! I'm getting the same problem with displaying S and P spots, but with additionally errorneous decoding. My device reproducing the same repeated test I2C frame:  addr 0x40 write, followed by sequence of four bytes: 0x01, 0x02, 0x03, 0x04. It can be seen in attached file that the first frame was decoded properly, but the next frames got errors - there are 7 dots in addr decode (1759.26 us), then 7 dots in the last byte 0x04 (2469,42us) and undecoded two bit intervals before the STOP. It's very annoying and I can't do my work with this instrument. Please fix it ASAP. The file attached was renamed to txt. Thanks!
[DSLogic-la-220521-001220 ERRORNEOUS I2C DECODING.txt](https://github.com/DreamSourceLab/DSView/files/8744698/DSLogic-la-220521-001220.ERRORNEOUS.I2C.DECODING.txt)
Another issue concerning displaying behaviour is that the Ack, Nack and Write events are 8 times narrower than data byte, but when I zoom out captured screen in order to view long transaction, I've got decoded data disappearing based on impossibility of your code to show narrow events, but there are still alot of space to show byte values. Please fix this too or make the "collapsing threshold" option in settings. Thanks!

_Originally posted by @BrainsLab in https://github.com/DreamSourceLab/DSView/issues/481#issuecomment-1133412645_

dreamsource-tai, 2022-05-24:
@BrainsLab 
Is your operation system linux or windows?

BrainsLab, 2022-05-24:
@dreamsource-tai 
On Windows-7 and Windows-10 were tested.

DreamSourceLab, 2022-05-25:
@BrainsLab 
It seems that this wave is not a valid I2C signal.
SDA signal should keep high in idle state.
[image]

BrainsLab, 2022-05-25:
@DreamSourceLab 
The SDA signal is probe number 0 and it goes idle state at around +0.8ms on this particular screenshot. There is narrow Idle interval between the STOP and START conditions on the bus. Probe #2 is just my debug IRQ signal.

DreamSourceLab, 2022-05-25:
@BrainsLab 
The error is caused by an invalid toggle near the rising edge of SCL
At around 1.8412ms
[image]

The frequency of SCL is only 50KHz, your sample rate(500MHz) is too high.
So multiple toggles are captured in one rising edge.
For 50KHz signal, 1MHz sample rate is enough.

BrainsLab, 2022-05-25:
@DreamSourceLab 
Thanks a lot! This was the reason. Now I found it too using your very helpful measure edges feature! Could you add to your to-do list some kind of visualization of such not obvious condition, for example, using the dots of red color, or attention sign on the timeline? In other words, some indication telling user there is something hidden from view because of wide zoom.

dreamsource-tai, 2022-05-26:
@BrainsLab 
OK, we have collected this demand. thank you.