DreamSourceLab/DSView #495: Problem compiling DSView at Mint LMDE5 (elsie)

tedy58, 2022-05-26:
I compile DSView few times in different Linux distributions (Ubuntu20.04, Debian 9,10,11 and Linux Mint LMDE4) recently I upgrade from Linux Mint LMDE4 to LMDE5 (elsie) and stack on lot of problems in my opinion it concern latest libboost1.74 library. seems there is some incompatibility with older libboost1.67 which works just like a charm and this latest lib. 
I reinstall all the libs as is described in the INSTALL file, use the DSView-master as source code. 
finally when try to make the compilation after lot of warnings I get the following result:

`[ 99%] Building CXX object CMakeFiles/DSView.dir/qrc_breeze.cpp.o
[ 99%] Building CXX object CMakeFiles/DSView.dir/qrc_language.cpp.o
[100%] Linking CXX executable DSView
/usr/bin/ld: /usr/local/lib/libpython3.9.a(dynload_shlib.o): undefined reference to symbol 'dlsym@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/DSView.dir/build.make:3007: DSView] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
`

I appreciate any help that I can obtain.

dreamsource-tai, 2022-05-27:
@tedy58 
Is your source code up to date? Can I see the cmake log? not is make log.

dreamsource-tai, 2022-05-27:
@tedy58 
Execute ".clean" first, then "cmake ."

tedy58, 2022-05-30:
**@dreamsource-tai** thank you for reply!
I tried with commands above, and Yes I do fresh clean installation of source with git clone, and I also do it with fresh downloaded master.zip code the same on all. I follow the steps as I did before, and new steps that include only `cmake .` and results always finish with that above. 
Now I think to dowload and latest python 3.10 and try with it to compile and will share results here.

dreamsource-tai, 2022-05-31:
@tedy58 
I would like to see the output of the execution cmake, which is helpful for analyzing the problem.
It is not the content you provided at the top.

tedy58, 2022-05-31:
**@dreamsource-tai** 
I will save the full log of compilation and will send it here.

tedy58, 2022-05-31:
tedy58@tedy58-5570:~/DSView$ ./clean 
clean cmake cache begin...
rm ./build
rm ./CMakeFiles
rm ./cmake_install.cmake
rm ./Makefile
rm ./Makefile.tmp
rm ./CMakeCache.txt
rm ./*.cmake
rm ./DSView.qrc.depends
rm ./moc_*.cpp*
rm ./qrc_*.cpp
rm ./install_manifest.txt
clean cmake cache end.
tedy58@tedy58-5570:~/DSView$ cmake . 
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'glib-2.0'
-- glib-2.0:
-- 	- includes:/usr/include/glib-2.0/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- 	- libraries:/usr/lib/x86_64-linux-gnu/libglib-2.0.*
-- Found PythonLibs: /usr/local/lib/libpython3.9.a (found suitable version "3.9.6", minimum required is "3") 
-- python3:
-- 	- includes:/usr/local/include/python3.9
-- 	- libraries:/usr/local/lib/libpython3.9.a
-- FFTW:
-- 	- includes:/usr/include
-- 	- libraries:/usr/lib/x86_64-linux-gnu/libfftw3.so
-- libusb-1.0:
-- 	- includes:/usr/include/libusb-1.0
-- 	- libraries:/usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- zlib:
-- 	- includes:/usr/include
-- 	- libraries:/usr/lib/x86_64-linux-gnu/libz.so
-- Checking for one of the modules 'libzip'
-- libzip:
-- 	- includes:
-- 	- libraries:/usr/lib/x86_64-linux-gnu/libzip.*
-- use Qt5:
-- 	- includes:/usr/include/x86_64-linux-gnu/qt5//usr/include/x86_64-linux-gnu/qt5/QtCore/usr/lib/x86_64-linux-gnu/qt5//mkspecs/linux-g++
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.42")  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tedy58/DSView
tedy58@tedy58-5570:~/DSView$ make 
[  0%] Generating qrc_language.cpp
[  0%] Generating DSView/moc_mystyle.cpp
[  0%] Generating DSView/pv/moc_sigsession.cpp
/home/tedy58/DSView/DSView/pv/sigsession.h:0: Note: No relevant classes found. No output generated.
[... 667 lines cut ...]
 1607 | #define _POSIX_C_SOURCE 200809L
      | 
In file included from /home/tedy58/DSView/libsigrokdecode4DSL/util.c:21:
/home/tedy58/DSView/libsigrokdecode4DSL/config.h:113: note: this is the location of the previous definition
  113 | # define _POSIX_C_SOURCE 200112L
      | 
[ 99%] Building C object CMakeFiles/DSView.dir/libsigrokdecode4DSL/version.c.o
In file included from /usr/local/include/python3.9/Python.h:8,
                 from /home/tedy58/DSView/libsigrokdecode4DSL/libsigrokdecode-internal.h:28,
                 from /home/tedy58/DSView/libsigrokdecode4DSL/version.c:21:
/usr/local/include/python3.9/pyconfig.h:1607: warning: "_POSIX_C_SOURCE" redefined
 1607 | #define _POSIX_C_SOURCE 200809L
      | 
In file included from /home/tedy58/DSView/libsigrokdecode4DSL/version.c:20:
/home/tedy58/DSView/libsigrokdecode4DSL/config.h:113: note: this is the location of the previous definition
  113 | # define _POSIX_C_SOURCE 200112L
      | 
[100%] Linking CXX executable build/DSView
/usr/bin/ld: /usr/local/lib/libpython3.9.a(dynload_shlib.o): undefined reference to symbol 'dlsym@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/DSView.dir/build.make:4200: build/DSView] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
tedy58@tedy58-5570:~/DSView$

tedy58, 2022-06-11:
I solve the problem by simply remove Linux Mint LMDE5 (elsie) and isntall Debian 11 bullseye.