DreamSourceLab/DSView #496: Why new changes are not getting released?

LordRafa, 2022-05-27:
Hi, I just seen that there are a lot of activity on this repo but no new releases has been made for a while I would like to know if there is any particular reason for this?

dreamsource-tai, 2022-05-27:
@LordRafa 
Before that, we have been developing another branch, which is now merged.

LordRafa, 2022-05-27:
Wow that is cool, I thought that this project was closed. I will stay tuned for the next release. Any clue when will be released?

dreamsource-tai, 2022-05-28:
@LordRafa 
The internal test is in progress, and a new version under windows will be released in the near future.

dreamsource-tai, 2022-06-23:
@tedy58 
This project uses python3. We developed it under Linux, so the code on GitHub is always up-to-date. On Linux, you can easily compile it.