DreamSourceLab/DSView #497: Linking issue due to executable name which is same as source directory

katyo, 2022-05-30:
I have an issue while compiling the master branch on Linux:
```
[100%] Linking CXX executable DSView
/bin/ld: cannot open output file DSView: Is a directory
```
Seems needs renaming or something else.

katyo, 2022-05-31:
@dreamsource-tai
I try build latest sources from master on NixOS. The output directory is 'build'.

katyo, 2022-05-31:
@dreamsource-tai 
Seems NixOS outputs cmake build artifacts into `build` directory by default.
When I changed `EXECUTABLE_OUTPUT_PATH` it links without issues.

dreamsource-tai, 2022-05-31:
@katyo 
If you change it to '', will it compile successfully?
Changed like set(EXECUTABLE_OUTPUT_PATH "${CMAKE_CURRENT_SOURCE_DIR}/build-bin") , it can compiled success?

katyo, 2022-05-31:
@dreamsource-tai 
Yes it links without issues.