DreamSourceLab/DSView #498: 新特性 建议

lclslsls, 2022-06-01:
很多时候需要抓很长的波形才能抓到感兴趣的部分，整体比较大，解码时间比较长，保存时间比较长。保存的时候能不能仅保存感兴趣的部分  比如 选择 curese1 到curese2 之间的波形才保存。

dreamsource-tai, 2022-06-02:
@lclslsls 
好的，谢谢你的建议。

hqs1, 2022-06-02:
> @lclslsls 
> 好的，谢谢你的建议。

你好，还有个建议在串口解析CMUX协议。
CMUX可能在串口复用用的比较多，一个硬件串口虚拟多个串口业务。

dreamsource-tai, 2022-06-06:
@hqs1 
因为开发工作量大，有的任务一时排不上来。但是这个项目是完全开源的，我们来开发协议的话，需要对各个协议的逻辑相当熟悉才能开发，这个过程比较耗时。用户就不一样，他们对协议的业务熟悉，时间也多。所以有的用户自己会去开发协议，有的也会贡献协议源码。你也可以试一下哦。
协议开发教程： https://wap.qidian.qq.com/ol/rest/view/3009108502_11402_1_1650351587

micky-git, 2022-07-15:
> @hqs1 因为开发工作量大，有的任务一时排不上来。但是这个项目是完全开源的，我们来开发协议的话，需要对各个协议的逻辑相当熟悉才能开发，这个过程比较耗时。用户就不一样，他们对协议的业务熟悉，时间也多。所以有的用户自己会去开发协议，有的也会贡献协议源码。你也可以试一下哦。 协议开发教程： https://wap.qidian.qq.com/ol/rest/view/3009108502_11402_1_1650351587

其实没有必要做范围存储，做自动打curser会比较有效，存盘文件存下打的curser的位置，打开文件的时候自动在对应位置上打上curser

lllaaa, 2022-08-07:
附议。保存和导出的时候都可以选择手工输入时间范围或者选择cursor之间数据会比较方便。kingst和acute的都有此功能。

dreamsource-tai, 2022-08-08:
@lllaaa 
这个需求目前是在我们的计划中的，下个版本会实现的。