DreamSourceLab/DSView #500: Logitech MX scroll wheel zoom doesn't work on MacOS

fishnux, 2022-06-07:
Scroll wheel zoom of some Logitech MX mouses don't work with DSView. It seems DSView is not handling certain mouse events

Same issue happens if one tries to use Mos for smooth scrolling:
https://mos.caldis.me/

So, if you want to test with a regular mouse, just download that program and see that mouse scroll doesn't work on DSView

dreamsource-tai, 2022-06-07:
@fishnux 
wave windows
a. normal mouse: wheel
b. magic mouse: shift + one Finger up&down on the mouse
c. touchpad: shift + two fingers up& donw on the touchpad
d. touchpad: pinch operations
osc vertical dial
a. normal mouse: hover + wheel
b. magic mouse: hover + one Finger up&down on the mouse
c. touchpad: hover + two fingers up& donw on the touchpad

fishnux, 2022-06-07:
Hi @dreamsource-tai 

Yes, I understand that. This particular mouse brand (some models) is not working with DSView, but does work with PulseView. Most likely, DSView is not handling certain mouse events. 

You can see one more person asking for same support:
https://github.com/DreamSourceLab/DSView/issues/311#issuecomment-727516097

You can see one more issue here:
https://github.com/DreamSourceLab/DSView/issues/479

I don't know why those issue creators closed the issue, all I can say is the problem still exists

dreamsource-tai, 2022-06-07:
@fishnux 
There are many models of this mouse,what is the specific model of your mouse?

fishnux, 2022-06-07:
@dreamsource-tai 

These are my mouses:
* Logitech MX Ergo
* Logitech MX Vertical

You can also use the application Mos with any standard mouse to see the same effect of scroll wheel not working on DSView for MacOS

itsonlyaudio, 2022-06-13:
It's also not working on Ubuntu Linux 22.04 with my Logitech Performance MX.

dreamsource-tai, 2022-06-14:
@fishnux 
We tried with logi MX ERGO, but DSView can workes. can you upload the pictrure of you mouse?

dreamsource-tai, 2022-06-14:
@itsonlyaudio 
And can you upload the pictrue of your mouse?

fishnux, 2022-06-14:
> @fishnux We tried with logi MX ERGO, but DSView can workes. can you upload the pictrure of you mouse?

Hi,

In my previous post I mistyped the model name (it's not MX Ergo), my apologies for the mistake. These are the mouses that I have tested and don't work on DSView on MacOS:
https://www.logitech.com/en-us/products/mice/lift-vertical-ergonomic-mouse.html
https://www.logitech.com/en-us/products/mice/mx-anywhere-3.html

Remember to install Logitech Options+:
https://www.logitech.com/en-us/software/logi-options-plus.html

This software smooths the scroll, so it's no longer discrete wheel events, it's a continuous scroll, that's why some mouses work and others don't

dreamsource-tai, 2022-06-14:
@fishnux 
If have not insalled the 'Logitech Options' software,  can the mouse works on DSView?
I means that can the mouse works on DSView without 'Logitech Options' software?

fishnux, 2022-06-14:
> @fishnux If have not insalled the 'Logitech Options' software, can the mouse works on DSView? I means that can the mouse works on DSView without 'Logitech Options' software?

That is correct, yes, just tested it

The issue can also be replicated with any mouse using the software Mos - which is another software that takes the discrete mouse wheel events and transforms it to "pixel precision" ones

dreamsource-tai, 2022-07-05:
@fishnux 
@itsonlyaudio 
Sorry, I will give you a final reply now.
If mouse works with mos software, must press shift key to zoom logic graph view.

fishnux, 2022-07-05:
PulseView works fine without any sort of workarounds like "press shift key", issue stands open

dreamsource-tai, 2022-07-07:
@fishnux 
We have released the new version v1.2.0, please download from "www.dreamsourcelab.com". It fix this issue.
Or you can compile it yourself.