DreamSourceLab/DSView #502: QSPI decider seems wrong

panpgg, 2022-06-09:
Hi ,
I have used DSView to decode QSPI waveform but it's seems wrong.
Flash is MX25R6435F, my seeting are
CLK : 0
IO0 : 1
IO1 : 2
IO2 : 3
IO3 : 4
CS : 5 (active low)
CPOL = CPHA = 0

I make sure I send 0x5A but I can't see 0x5A in waveform decode.
Is there anything I missed ?

Best Regards,
[DSLogic-la-220609-092658.zip](https://github.com/DreamSourceLab/DSView/files/8866003/DSLogic-la-220609-092658.zip)

DreamSourceLab, 2022-06-10:
@panpgg 
What is the clock frequency of QSPI? 
The sample rate is too low, so the real signal is not sampled correctly. 
We recommend a sample rate of 10 times than under test singals, at least 4 times.

panpgg, 2022-06-10:
@DreamSourceLab 
The clock frequency of QSPI is 2MHz. I think it's enough but I am going to increase sample rate and test again.
Are there other settings or HW issues to caused this issue ?

DreamSourceLab, 2022-06-10:
@panpgg 
But in your dsl file, the frequency is not 2MHz.
There are many pulse up to 12.5MHz.