DreamSourceLab/DSView #506: 新特性建议：自动保存

lclslsls, 2022-06-30:
在长时间测试的时候，有时候需要连续几十个小时的测试。现在的存储深度跟不上这种无底洞的。
可不可以每隔固定时间（例如一小时）自动保存，事后可以方便分析。

micky-git, 2022-07-14:
> @lclslsls 我们先收集这个需求，以后版本加上去。

这个确实是一个痛点，我自己写了一个循环自动保存和自动从新开始采集，但是咱们存盘不支持多线程每次只存一个buffer的模式存盘时间非常慢，所以我自己又写了一个独立线程存盘，发起存盘立刻复制一整份buffer然后丢给独立的线程去存盘，立即从新开始采集，这样能采集到时间上基本连续的数据，大规模测试非常需要连续性。所以建议存盘和采集能分立，连续采集，采集的同时存上一轮的数据。

micky-git, 2022-07-14:
我测试了一下，写成多线程每个probe独立存一个zip，最后再拼合成一个zip能取得6-7倍左右的多线程加速（amd 3400g 4核8线程）

dreamsource-tai, 2022-07-14:
@micky-git 
你用的是什么系统呢？

micky-git, 2022-07-15:
> @micky-git 你用的是什么系统呢？

Ubuntu下1.12的DSVIEW，之前咱们更新比较少，反向搜索不了，存盘慢这些问题只能自己改代码解决。新版本我看更新日志里面这些问题解决了。新版的minizip是支持多线程了吗？现在1.2版本的DSVIEW存盘速度我还没有尝试。

dreamsource-tai, 2022-07-18:
@micky-git 
新版的数据存储是先在内存中生成zip文档的，所以最好是一个线程去操作。而且速度比之前快。
如果你想多个线程存储，估计是每个线程只能操作一个zip文档，然后再合并了。
而且我们在进行大的重构，后续存储完全由libsigrok4DSL库完成，只需要给它一个文件路径就完成保存。