DreamSourceLab/DSView #510: 如何编译出在windows上运行的软件

micky-git, 2022-07-14:
需要在linux下交叉编译吗？还是可以在windows下直接编译出exe。

dreamsource-tai, 2022-07-15:
@micky-git 
windows下编译很麻烦，建议直接下载编译好的程序。
下载地址： www.dreamsourcelab.cn

micky-git, 2022-07-15:
> 

DSView是一个非常好的浏览软件但是他在一些专业领域（激光测量）缺少很多实用的功能，我针对自己的需要改了很多代码（统计功能，自动化的分析和条件存储，后台高速解码，质量分析）这些在linux下运行效果很好，希望可能迁移到win下。

RedConda, 2022-07-15:
可以一起交流交流啊，我也想在win下把软件编译出来

DreamSourceLab, 2022-07-15:
@micky-git 
DSView是一个开源软件，如果有修改请保持开源并commit给我们，我们会评估并合并相应的features。

micky-git, 2022-07-22:
> @micky-git DSView是一个开源软件，如果有修改请保持开源并commit给我们，我们会评估并合并相应的features。

我的网络实在是传不上去github，有什么其他渠道提交吗

DreamSourceLab, 2022-07-23:
> > @micky-git DSView是一个开源软件，如果有修改请保持开源并commit给我们，我们会评估并合并相应的features。
> 
> 我的网络实在是传不上去github，有什么其他渠道提交吗

优先github，实在不行可以发送到support@dreamsourcelab.com

micky-git, 2022-07-26:
> @micky-git windows下编译很麻烦，建议直接下载编译好的程序。 下载地址： [www.dreamsourcelab.cn](http://www.dreamsourcelab.cn)

能不能给点提示，是用msys2 还是用什么方法，比较麻烦是指在win平台原生编译配环境吗？我加入的一些特性是专用的不太可能通过评估合并入DSView。

sagasm, 2022-09-19:
@micky-git 我用msys64编译成功，但是整个过程确实很麻烦，特别是libusb库，在windows只能用特定版本。

WBF001, 2023-05-25:
> @micky-git 我用msys64编译成功，但是整个过程确实很麻烦，特别是libusb库，在windows只能用特定版本。

请问需要哪个特定版本

nrzo, 2023-08-14:
采用如下步骤使用msys2能成功编译，但是很遗憾：硬件连接DSLogic点击start后崩溃

1. pacman -S mingw-w64-x86_64-make mingw-w64-x86_64-cmake mingw-w64-x86_64-gcc mingw-w64-x86_64-glib2 mingw-w64-x86_64-qt5-tools mingw-w64-x86_64-qt5-winextras mingw-w64-x86_64-libusb mingw-w64-x86_64-boost mingw-w64-x86_64-fftw mingw-w64-x86_64-python3 mingw-w64-x86_64-libtool mingw-w64-x86_64-zlib
2. cp /mingw64/bin/mingw32-make.exe /mingw64/bin/make.exe
3. /mingw64/bin/cmake.exe CMakeLists.txt -G"Unix Makefiles"
4：modify CMakeLists.txt:
    find_package(Qt5WinExtras REQUIRED)
    set(QT_INCLUDE_DIRS ${Qt5Gui_INCLUDE_DIRS} ${Qt5Widgets_INCLUDE_DIRS} ${Qt5WinExtras_INCLUDE_DIRS})
    set(QT_LIBRARIES Qt5::Gui Qt5::Widgets Qt5::WinExtras)
    #set(QT_INCLUDE_DIRS ${Qt5Gui_INCLUDE_DIRS} ${Qt5Widgets_INCLUDE_DIRS})
    #set(QT_LIBRARIES Qt5::Gui Qt5::Widgets)
5. make -j32
6. Add Environment Variable(modify ~/.bashrc)：
    export PYTHONHOME="/mingw64"
    export PYTHONPATH="/mingw64/bin"
    export QT_QPA_PLATFORM_PLUGIN_PATH="/mingw64/share/qt5/plugins/platforms"
    export PATH=/mingw64/bin:$PATH
7. run DSView.exe
    cd build.dir
    ./DSView.exe

nrzo, 2023-08-14:
> @micky-git 我用msys64编译成功，但是整个过程确实很麻烦，特别是libusb库，在windows只能用特定版本。

你好！请问你编译完成后运行正常吗？我按您的方法采用msys64编译成功，但是运行点击start崩溃：（

lclclc666789, 2024-07-24:
我整了个 vs studio 编译的dsview ，各位大佬试试 https://github.com/lclclc666789/DSView

RedConda, 2024-07-25:
> 我整了个 vs studio 编译的dsview ，各位大佬试试 https://github.com/lclclc666789/DSView

这个真的可以吗，我感觉msys2编译出来的都不能用

lclclc666789, 2024-07-25:
肯定可以啊，改天我录个视频

WHG555, 2025-05-21:
windows 7可以编译么