DreamSourceLab/DSView #515: CUDA添加问题

micky-git, 2022-07-26:
在CMakelists中添加CUDA编译时NVCC报错：
添加内容：
project(DSView LANGUAGES C CXX CUDA)
set (CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11")

似乎是因为
add_definitions(xxx)中添加的内容

注释掉：
#add_definitions(${QT_DEFINITIONS})
#add_definitions(-Wall -Wextra -Wno-return-type -Wno-ignored-qualifiers)
报错：nvcc fatal   : redefinition of argument 'optimize'
add_definitions是全局生效吗？（对C C++ CUDA都添加了相同的参数）

dreamsource-tai, 2022-07-27:
@micky-git 
不太理解你的问题，你是想链接一个库，还是什么？

micky-git, 2022-07-27:
> @micky-git 不太理解你的问题，你是想链接一个库，还是什么？

想给DSView添加对异构计算CUDA的支持，添加了一段cuda代码
set(cuda_SOURCES
DSView/test.cu
)
但是我不太熟悉cmake配置，单独编译能够运行，添加到DSView的cmakelists里面混合编译没成功

micky-git, 2022-07-27:
> @micky-git 不太理解你的问题，你是想链接一个库，还是什么？

DSView很适合做验证工作，比如给10亿个重复的信号，通过解码分析误码率。用来验证仪器的可靠性。
DSView很可以做统计工作，比如画出时钟的cnt的正态分布，计算方差等
不过目前解码的时间比较长（秒级到分钟级别），在计算上有瓶颈。比如目前统计500万个边沿的计数和占空比大概需要0.3秒
DSView核心算法效率已经很高没有多少优化空间
后面再提速的方式：
1.多线程
2.avx
3.GPU异构计算
多线程是最简单直接的（加速倍数看核心数），GPU计算应该是最快的（*1000）

dreamsource-tai, 2022-07-28:
@micky-git 
有的协议解码，在数据量大的时候会占用不少时间，后面我们会对一些耗时的协议转成c代码解析。

micky-git, 2022-07-28:
> @micky-git 有的协议解码，在数据量大的时候会占用不少时间，后面我们会对一些耗时的协议转成c代码解析。

DSView如果添加对CUDA计算的支持，会有很多现成的算法模型可用，相比于Logic 2 插件模式，DSView专业性能上更有潜力，尤其是精密仪器的统计验证，非常耗时，CUDA并行计算大概有1600倍的加速。

dreamsource-tai, 2022-07-28:
@micky-git 
好的，谢谢你的建议，等后边我们再着手调研一下。

micky-git, 2022-07-28:
> @micky-git 好的，谢谢你的建议，等后边我们再着手调研一下。

找到解决方案了
https://gitlab.kitware.com/cmake/cmake/-/issues/16915
原因是qt5和cuda编译参数冲突

cmakelist.txt:
...
if ( TARGET Qt5::Core )
	get_property( core_options TARGET Qt5::Core PROPERTY INTERFACE_COMPILE_OPTIONS )
	string( REPLACE "-fPIC" "" new_core_options "${core_options}" )
	set_property( TARGET Qt5::Core PROPERTY INTERFACE_COMPILE_OPTIONS ${new_core_options} )
	set_property( TARGET Qt5::Core PROPERTY INTERFACE_POSITION_INDEPENDENT_CODE "ON" )
	set( CMAKE_CXX_COMPILE_OPTIONS_PIE "-fPIC" )
endif()
...

micky-git, 2022-07-28:
DSView: Switch to device: "Demo Device" 
DSView: Hotplug thread start!
_____________________________#info#_________________________________ 
使用GPU device 0: NVIDIA GeForce RTX 3070 Laptop GPU
全局内存量MB 8191
block中可用32位寄存器数量 65536 
warp块大小 32
每个block中最大线程 2147483647
时钟频率 1560000 
显存主频 7001000
SM的数量：40
gridsize：1560000
每个线程块的共享内存大小：48 KB
每个线程块的最大线程数：1024
每个EM的最大线程数：1536
每个SM的最大线程束数：48 
_____________________________#info#_________________________________ 
_cpu_decode() time: 27.468ms
_gpu_decode() time: 0.059ms
Max error: 0

WSL2中 计算1<<25=30M elements大概有500倍加速比

sagasm, 2022-09-19:
@micky-git 请问现在的python解码器可以用cuda么

dreamsource-tai, 2022-10-08:
@sagasm 
协议没有的话，你要自己写，或者从别的地方下载后加进去。存在兼容问题的话，需要改一下代码。