DreamSourceLab/DSView #516: Wrong triggering on I2C when Repeated Start exists

Krapet, 2022-07-27:
It looks that described issue is probably related to **Repeated Start** condition.
Any bit combination in Trigger Setting before Repeated Start works fine and also trigger works as expected.
After Repeated Start condition it has troubles to get matched trigger condition.

My packet starts with **writing** to address 0x10, so first byte after start condition is 0x20.
Then packet continues with some data bytes.
After restart condition the **reading** is requested from address 0x10, so first byte is 0x21.
Then packet continues with some other data bytes.

The first byte is not relevant for described issue. Shown just for comparison of Start and Repeated Start condition.
[image]

In "The trigger setting" the "Data Value" is: X X X X X X X 0 0 1 0 0 0 0 1 X
As my opinion this represents 0x21 (requested) and ACK (not checked).

I expect that this Trigger setting catch condition (at time 2455us). Unfortunately it doesn't.
[image]

It triggers on data of value 0x21. (at time 5020us). And probably not always.
[image]

As you can see, the data bits on both picture are the same. 
Unfortunately it triggers much later than expected.

DSView Settings:
[image]

An finally. 
Repeatable state of non working trigger is:
- Packet structure:
  Start Condition, 
  Address byte + Write, 8 data bytes,
  Repeated Start, 
  Address byte + Read, 5 data bytes (0xFF, 0xFF, 0xFF, 0xFF, 0xFF),
  Stop Condition.
- Trigger Settings:
  Data Value:  X X X X X X X 1 1 1 1 1 1 1 1 X
  (to catch byte 0xFF)
**This is NEVER TRIGGERED !!**

My device is DSLogic U3Pro16.

DreamSourceLab, 2022-08-08:
@Krapet 
According to the principle of Serial Trigger, Repeated Start will NOT be treated as a Start Flag, because there is no Stop  Flag before it.
We will try some new implements in future version, for example recognize Start Flag even without Stop Flag. This maybe fullfill your requirment on Repeated Start case.

Krapet, 2022-08-19:
I agree that Start and Repeated Start may be treated as different flags. 
I2C definition of S and Sr is: Falling edge on SDA while SCL is high. It is not conditioned by Stop flag.
Ambiguous to me is state when triggered is last byte in packet (of value 0x21) and not byte after Sr (also 0x21).
I hope this helps to make useful changes in I2C demux.