DreamSourceLab/DSView #517: WSL2 ezusb: failed to open device: LIBUSB_ERROR_ACCESS.

micky-git, 2022-07-28:
在window中开启WSL2安装ubuntu
在通过usbipd挂载给ubuntu

> lsusb
Bus 002 Device 002: ID 2a0e:002a DreamSourceLab USB-based DSL Instrument v2
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 
成功挂载后

>DSView
DSView: ----------------- version: 1.2.0----------------- 
DSView: Qt:5.12.8 
DSView: decode script path: "/usr/local/share/libsigrokdecode4DSL/decoders"
sr: ezusb: uploading firmware to device on 2.2                                                                          
sr: ezusb: failed to open device: LIBUSB_ERROR_ACCESS. 
sr: dslogic: Firmware upload failed for device 0.   
DSView: Switch to device: "Demo Device" 
DSView: Hotplug thread start! 

>sudo DSView
会卡死

dreamsource-tai, 2022-07-29:
@micky-git 
先把硬件拿掉，然后运行软件，看界面能不能出来。

micky-git, 2022-07-29:
> @micky-git 先把硬件拿掉，然后运行软件，看界面能不能出来。

在没有设备的情况下sudo DSView 可以正常运行
>
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
DSView: ----------------- version: 1.2.0-----------------
DSView: Qt:5.12.8
DSView: decode script path: "/usr/local/share/libsigrokdecode4DSL/decoders"
DSView: Switch to device: "Demo Device"
DSView: Hotplug thread start!           

软件处于运行状态再插入usb
在win11 cmd下通过usbipd把windows的usb共享给wsl2 ubuntu
此时DSView没有反应，还是demo device