DreamSourceLab/DSView #520: 新特性建议：为DSCope中的FFT添加Markers功能

GengKanghua, 2022-08-13:
希望添加多个Marker关注多个特定频点的幅度。
具体的可以参考NI InstrumentStudio 2022软件的功能说明
https://www.ni.com/docs/zh-CN/bundle/instrumentstudio-22q2-feature/page/fft-peaks-markers.html

dreamsource-tai, 2022-08-15:
@GengKanghua 
能说具体点吗？

GengKanghua, 2022-08-15:
> @GengKanghua 能说具体点吗？

该需求更像是一个频谱仪的功能。当前，在测量时只能看到鼠标所在点的频率和幅度，但是我们有时需要关注多个特殊频点，这个需求用鼠标操作很难实现，第一，鼠标只能关注一个频点；第二，具体的频率鼠标并不能指示的很准确。
下图是一个频谱测量的场景，其中图片下部的表格里MR，M1，M2，M3的频点可以手动设置，设置好之后，表格最右侧就可以实时显示其数据，方便观测和对比。
[image]

补充：我观察了一下现有台式示波器，它的FFT里并没有此项功能。但是与台式示波器相比，DSView的载体是性能强大的PC，理应有更丰富的功能。
[image]

dreamsource-tai, 2022-08-15:
@GengKanghua 
好的，谢谢提出建议。我们经过评估后加入开发计划。