DreamSourceLab/DSView #525: Getting error when using the Numbers and State Decoder.

rfindlay68, 2022-08-22:
I'm trying to use DSView to decode 16 data inputs to a hex number (the address bus for a 6502) using the Numbers and states decoder.
It seems no matter how I configure the decoder for bits, etc, I only get an error shown in the attached file.
Can anyone help please?

[image]
?

dreamsource-tai, 2022-08-23:
@rfindlay68 
Please upload your test data and we will use it to reproduce the problem. Thank you.

dreamsource-tai, 2022-08-23:
@rfindlay68 
What decoder is it?

rfindlay68, 2022-08-23:
Good morning!  It is the Numbers and State Decoder.  I'll attach what I 
believe is the data to this reply.

If this is not the data you need please let me know how to export the date.

Regards, Richard.

rfindlay68, 2022-08-23:
This is another reply on the same issue #.

Good morning!  It is the Numbers and State Decoder.  I'll attach what I 
believe is the data to this reply.

If this is not the data you need please let me know how to export the date.

Regards, Richard.

dreamsource-tai, 2022-08-24:
@rfindlay68 
On the toolbar, click menu: "File" -> "Save".
Save the data to file,  you can reply this issue by email and attach the data, I will receive it.

rfindlay68, 2022-08-28:
Good evening!  I'd sent the data a couple of times.  Did you receive it ok?
Thank you.

dreamsource-tai, 2022-08-29:
@rfindlay68 
I haven't received the data you sent.
Send the data to me through the attachment of the mail please. My email id is "tzz@dreamsourcelab.com".

rfindlay68, 2022-08-31:
Good evening, please find the files that I was able to save attached.

Regards, Richard.

dreamsource-tai, 2022-09-07:
@rfindlay68 
[image]

dreamsource-tai, 2022-09-07:
@rfindlay68 
 Rename the dsl file name as zip file first.

rfindlay68, 2022-09-07:
I've Zipped the DSL file and attached it here in github.  Thank you.

[DSLogic-la-220822-091357.zip](https://github.com/DreamSourceLab/DSView/files/9509220/DSLogic-la-220822-091357.zip)

closed 2022-09-07 by rfindlay68

dreamsource-tai, 2022-09-08:
[pd.zip](https://github.com/DreamSourceLab/DSView/files/9525678/pd.zip)

dreamsource-tai, 2022-09-08:
@rfindlay68

dreamsource-tai, 2022-09-08:
@rfindlay68

dreamsource-tai, 2022-09-08:
@rfindlay68 
I sended you the decoder script file, please extract it and copy file "pd.py" to the directory "decoders/numbers_and_state", repleace the old file.If you have any questions, please give me feedback. Thank you.
[pd.zip](https://github.com/DreamSourceLab/DSView/files/9525722/pd.zip)

rfindlay68, 2022-09-09:
Thank you for sending the file. I'll try it out and let you know.  Take care.

rfindlay68, 2022-09-10:
Good morning!  I just tried the new file for the Numbers and State 
decoder and it seems to work properly.  I now get good decoding on the 
data busses.  Thank you very much for your help.

Regards, Richard.