DreamSourceLab/DSView #531: U3Pro16 sampling time inaccurate

roberto-mainetti, 2022-09-05:
I'm reading a UART signal with configuration 115200 8N1, using a U3Pro16 device in buffer mode.
The data I'm sending is number 68.0F and 50.5F, that translate into 0x00008842 and 0x00004A42.
The sampling configuration is: buffered, 1MHz, 1.00s, but to have the UART decoder translating the signal to the correct reading I have to set it to 138000 8N1, not 115200 8N1:
[image]
But if I set the configuration to: buffered, 100MHz, 1.00s then the conversion time is messed up again:
[image]
So at this point I would have to change the decoder baud rate configuration to 160000 to get a proper conversion:
[image]

The timings should not change with the sample rate.

I suspect the device cannot keep up with the sampling, or something like that, but it's odd, being that it should be capable of reaching 1GHz.

Similar issues happen if I use the "Stream mode" instead of the "Buffered mode".

roberto-mainetti, 2022-09-05:
I need to add that this issue is happening on a Windows 10 PC running DSView 1.2.1, I have tried a PC running Ubuntu and DSView 1.2.0, and in there it seems to work properly, reading the serial data with configuration 115200 8N1, sampling at 100MHz.

DreamSourceLab, 2022-09-06:
@roberto-mainetti 
The acquisition accuracy is determined by the sampling rate.
For 1M sample rate, signal will be sampled each 1000ns.
For 100M sample rate, singal wll be sampled each 10ns.
If you want to test frequency accuracy, you can find a reference signal generator.
Intuitively, it is unlikely that the two waveforms you captured are the same signal. 
You can upload the dsl file, and we will do further analysis.

roberto-mainetti, 2022-09-06:
@DreamSourceLab

Thank you for your answer. Here attached you can find the acquisition of the same signal as before, periodically repeated. The decoding of this UART signal requires a baud rate of 122000. The acquisition was done at 100 MHz, Buffered mode, using a PC running Windows 10 and using DSView 1.2.1.

[DSLogic-la-220906-143518.zip](https://github.com/DreamSourceLab/DSView/files/9496930/DSLogic-la-220906-143518.zip)

The following instead is an acquisition of the same signal, without touching anything on the signal generator side, I just disconnected the U3Pro16 from the Windows PC and connected it to the Ubuntu PC, and did the same acquisition with the same configuration. In this case I have the signal correctly decoded at baud rate 115200.

[DSLogic-la-220906-145225.zip](https://github.com/DreamSourceLab/DSView/files/9497058/DSLogic-la-220906-145225.zip)

In both cases the USB port used is a USB 3.0, directly connected to the computer, without HUBs in the middle, with the original USB cable.

On the Windows PC I have tried also other USB ports with similar results.

roberto-mainetti, 2022-09-06:
An update, I think it might be an issue with the driver that is selected by default, I have tried updating the driver, and selecting this seems to fix the acquisition:
[image]
There were other drivers on this PC, maybe from an installation of PulseView, used with another logic analyser.

DreamSourceLab, 2022-09-07:
@roberto-mainetti 
Which driver did you use before?
Could you confirm whether the firmware is provided by DSView or not?

roberto-mainetti, 2022-09-07:
If a driver different than the one provided by DS-View was installed it was not intentional for sure. I started looking into the drivers only after seeing the device behaving wrongly as described at the beginning of this thread.

Looking at the available drivers on this computer I can see these:
[image]
The first one is the one in the previous screenshot.
I went inside DSView installation path and tried to run dpinst-amd64.exe in Drivers folder, I had this feedback from this installer:
[image]
This action actually changed the driver in use by the device to this:
[image]
Interesting to notice, also this driver is working properly.

The driver that was in use when the instrument was not working properly is the last one listed in the screenshot about available drivers, that is WinUSB Generic Device Version 6.1.7600.16385.