DreamSourceLab/DSView #532: 截断保存

lengyechen, 2022-09-08:
hello ALL，
    DSview 有计划做截断保存吗？比如抓到的包太大，想从某一个点开始保存到另外一个时间点，很需要这个功能吗？有准备做这个功能吗？

dreamsource-tai, 2022-09-09:
@lengyechen 
能说具体点吗？

lengyechen, 2022-09-13:
[image]
比如文件已经抓取完成之后，重新打开文件可以在箭头的位置将文件分开保存成两份文件类似这样的功能。

我们会有这样的功能吗？ 谢谢

lengyechen, 2022-09-13:
很多场景可能需要这样的功能，例如1 文件过大，不好查找问题点的时候，可能之前抓到的波形并不需要，我可以自己操作将这个波形从某一个位置分成两份或者多份保存。 例如2 给别人share这个问题的时候，要找问题发生点的时候可能也会比较困难，需要重新去找问题点，所以如果文件能被截断的话比较方便看问题发生的点。

dreamsource-tai, 2022-09-14:
@lengyechen 
我们有计划实现保存文件时可指定时间范围的功能，不过你说的把它分成2个以上的文件，这不太合理，会让问题复杂，不过你可以通过多次保存来实现。

lengyechen, 2022-09-15:
“我们有计划实现保存文件时可指定时间范围的功能”  这个和我说的这个功能不一样吗？ 那我指定从箭头那里开始，到最后时间结束，还是说还是有点不一样的地方？

lllaaa, 2022-09-18:
建议实现截断保存的时候也可以一并选择哪些通道要被保存。保存时间可以基于任意cursor或者开始、末尾。