DreamSourceLab/DSView #533: LPC decode error

ksibium, 2022-09-09:
LPC decde with error message: object Decode have no attribute oldlframe.

dreamsource-tai, 2022-09-11:
@ksibium 
Please save data to file, and rename the file name as .zip, upload it to github.

dreamsource-tai, 2022-09-11:
@ksibium 
[image]

ksibium, 2022-09-28:
[DSLogic-la-220908-191953_100.zip](https://github.com/DreamSourceLab/DSView/files/9666790/DSLogic-la-220908-191953_100.zip)
problem in LPC decoders pd.py file. Mistakes in reset function.

ksibium, 2022-09-28:
lost parameter "self.oldlframe = -1"

dreamsource-tai, 2022-10-08:
@ksibium 
Yes,you can append a line of code such as "self.oldlframe = -1" to the reset() function.