DreamSourceLab/DSView #542: LIN bus decoder is mised up with UART:1 decoder

f-steff, 2022-10-02:
Mixup 1:
Add a few UART:1 and a few LIN decoders to a project. Store the configuration. 
Some of the LIN decoders are converted to UART:1 decoders.

Mixup 2:
Add a few UART:1 and a few LIN decoders to a project. Store the configuration. 
Some of the LIN decoders sometimes disappear.
This also results is an error similar to  "LIN is not subscribable" in red text on top of the decoder view.

dreamsource-tai, 2022-10-08:
@f-steff 
What is your DSView version?

f-steff, 2022-11-24:
@dreamsource-tai 
Apologies for the late reply. I'm on the latest downloadable version: v1.2.1. Tested primarily on Windows 10 64bit, and very briefly on some MacOS, too, just to confirm the error was also there.

To further elaborate.
Exactly what mixup/error occurs is not consistent. Sometimes one change occurs, and other times other changes occur. 
Storing and restoring a configuration especially seems to trigger the problem, although it appears just using the configuration and removing/adding to it can trigger it, too.

dreamsource-tai, 2022-11-25:
@f-steff 
If you add a sub decoder, its parent decoder will be automatically associated, so you do not need to add a parent decoder.

shidong-z, 2023-09-22:
In the LIN protocol, there is a problem with the PID resolution, the seventh and eighth bits are the check bit, and the check bit appears later. 
[image]