DreamSourceLab/DSView #543: Decoder channel names do not save with

JoshMorong, 2022-10-03:
I'm using:

DSView 1.1.2 (x64)
Website: [www.DreamSourceLab.com](https://github.com/DreamSourceLab/DSView/issues/www.DreamSourceLab.com)
Github: https://github.com/DreamSourceLab/DSView
Copyright:[© DreamSourceLab. All rights reserved.](https://github.com/DreamSourceLab/DSView/issues/%C2%A9%20DreamSourceLab.%20All%20rights%20reserved.)

On a Windows 10 machine.  

File>Settings>Save does not store the names I have given to decoder channels.  This limits the utility of the software because with multiple decoder channels monitoring different SPI and I2C channels, it is confusing when to interpret when this information is discarded.

Regards,
Josh

dreamsource-tai, 2022-10-08:
@JoshMorong 
Please download the latest version of the program. 
Website:[ www.dreamsourcelab.com](url)

JoshMorong, 2022-10-10:
I downloaded the latest version of the program, and the problem still exists.  The File save doesn't save the decoder channel names, and neither does the save->configuration.

Also, I had added a decoder by copying and renaming the 0-spi decoder, and adding code so that the data could be displayed in various formats because I am using an SPI to output ASCII messages, and that option wasn't available.  While doing that, I added in the other options available in UART formatting.  When I updated the software, this folder was deleted, and luckily I had a backup.  How do I work around this so that I don't lose any decoders I add in the future?

dreamsource-tai, 2022-10-11:
@JoshMorong 
On windows, when you reinstall the DSView program, the old file will be replaced, so your must to backup them.
What is your operating system language?

JoshMorong, 2022-10-11:
Thanks.  In this case, it's moot as you had updated the decoders so that all have formatting options--very nice!

I'm using Windows 10 in English.  

BTW, I find the cursor implementation very frustrating because I am often unable to pick a point on a waveform to measure against other channels.  It works about 40% of the time I need it too, and otherwise tries to put up a straight line with perpendicular lines at either end, that doesn't provide an obvious utility.

This is what I'm trying to do:
[image]

This is what I often get:
[image]

Any advise on how to reliably get the former option?

Thanks for your help.
Josh

dreamsource-tai, 2022-10-12:
@JoshMorong
The cursor problem: Your cursor needs to move closer to the edge.
Another question: Can you able to save the channel name now?

JoshMorong, 2022-10-13:
No, the decoder-channel names do not save with either Config-Save, or with File-Save.  As soon as I reload the configuration, or open the saved file, the decoder-channel names have reverted to their default names.

dreamsource-tai, 2022-10-20:
@JoshMorong 
Can you upload the saved file with custom channel name? Rename the file as .zip file first.

dreamsource-tai, 2022-10-20:
@JoshMorong 
Is your channel name as shown in the picture?

JoshMorong, 2022-10-20:
No problem.  Here's a screenshot of a file that I've renamed one of the Decoder-Channels from: "0: SPI" to "DEBUG:SPI" and am saving the file and the Configuration.  After I load the config, or open the file, the name will be "0:SPI" again.
Regards,
Josh

[image]
[RenamedDecoderChannel.zip](https://github.com/DreamSourceLab/DSView/files/9830088/RenamedDecoderChannel.zip)
[DecoderChannelRenamedConfig.zip](https://github.com/DreamSourceLab/DSView/files/9830089/DecoderChannelRenamedConfig.zip)

dreamsource-tai, 2022-10-21:
@JoshMorong 
I received your file, and load it, the channel name can be loaded well.
Please try: 1. Use the demo device to generate a small amount of data, name some of the notifications, and reload them to see if they are normal; 2. Record the log and send it to us. Log function: click the help menu on the top right.

dreamsource-tai, 2022-10-21:
@JoshMorong
Sorry, I didn't understand you correctly before. DSView currently only supports the naming of channels, but does not implement the naming of decoder information lines. We will implement this function in the new version. thank you.