DreamSourceLab/DSView #547: Python automation with Dslogic

Decee1, 2022-10-13:
Hello,

I want to do some automation testing with DsLogic/DsView
But is there a way of doing this?

dreamsource-tai, 2022-10-14:
@Decee1 
Can you be more specific?

Decee1, 2022-10-14:
Hello @dreamsource-tai 

I want to be able to read data from the DsLogic using Python so i can get for example periods and frequencies from when the trigger level was reached.

Another thing would be to access decoders that could decode the SPI, I2C and so on which is not required yet. 

I found that sigrok kind of can do this already but it requires more work to read periods, decode data and so on.
https://sigrok.org/wiki/DreamSourceLab_DSLogic
As it is not very.. Python friend as-is if you have to change a lot of parameters. 

So in a nutshell 
Be able to read data from the DsLogic with Python and output the read data so it can be decoded using decoders

dreamsource-tai, 2022-10-20:
@Decee1 
Do you mean the following explanation:
1. DSView provides the sdk library, which you can call through python to achieve data collection;
2. Python can access the data results decoded by DSView for further analysis;
DSView itself can collect data and decode it using python scripts. These decoder files are in the "decoders" directory. If you want to implement your own decoding rules, you need to write your own python code.

Decee1, 2022-10-21:
@dreamsource-tai Hi,

Yeah kind of. So if DSView supports those features it should be implemented into a Python package making it easier to retrieve the datas and decode it instead of people having to make their own package every time they need to measure data using DSView