DreamSourceLab/DSView #548: Link error for Ubuntu 18.04

tysenmoore-xse, 2022-10-14:
I pull the master and attempted a build. Everything builds until it gets to linking DSView. I get the following:
```
[100%] Linking CXX executable build.dir/DSView
CMakeFiles/DSView.dir/DSView/main.cpp.o: In function `main':
main.cpp:(.text.startup+0x25f): undefined reference to `LangResource::Instance()'
main.cpp:(.text.startup+0x26b): undefined reference to `LangResource::Load(int)'
CMakeFiles/DSView.dir/DSView/pv/mainwindow.cpp.o: In function `pv::MainWindow::switchLanguage(int)':
mainwindow.cpp:(.text+0x1685): undefined reference to `LangResource::Instance()'
mainwindow.cpp:(.text+0x1690): undefined reference to `LangResource::Load(int)'
CMakeFiles/DSView.dir/DSView/pv/toolbars/samplingbar.cpp.o: In function `pv::toolbars::SamplingBar::retranslateUi()':
samplingbar.cpp:(.text+0x129): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x14b): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
samplingbar.cpp:(.text+0x1c2): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x1dd): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
samplingbar.cpp:(.text+0x1e5): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x200): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
samplingbar.cpp:(.text+0x208): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x223): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
samplingbar.cpp:(.text+0x22b): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x246): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
samplingbar.cpp:(.text+0x335): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x350): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
samplingbar.cpp:(.text+0x3ae): undefined reference to `LangResource::Instance()'
samplingbar.cpp:(.text+0x3c9): undefined reference to `LangResource::get_lang_text(int, char const*, char const*)'
collect2: error: ld returned 1 exit status
CMakeFiles/DSView.dir/build.make:4412: recipe for target 'build.dir/DSView' failed
make[2]: *** [build.dir/DSView] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/DSView.dir/all' failed
make[1]: *** [CMakeFiles/DSView.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
```
I have attempted the dev dependedcies and they seem to be met. Any suggestions would be much appreciated.

tysenmoore-xse, 2022-10-14:
Got this to compile by fixing the make file:

diff --git a/CMakeLists.txt b/CMakeLists.txt
index b8c3072..9e28684 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -337,6 +337,7 @@ set(DSView_SOURCES
     DSView/pv/dock/protocolitemlayer.cpp
     DSView/pv/ui/msgbox.cpp
     DSView/pv/ui/dscombobox.cpp
+    DSView/pv/ui/langresource.cpp
     DSView/pv/dsvdef.cpp
     DSView/pv/dialogs/applicationpardlg.cpp
     DSView/pv/dock/keywordlineedit.cpp
@@ -428,6 +429,7 @@ set(DSView_HEADERS
     DSView/pv/dock/protocolitemlayer.h
     DSView/pv/ui/msgbox.h
     DSView/pv/ui/dscombobox.h
+    DSView/pv/ui/langresource.h
     DSView/pv/dsvdef.h
     DSView/pv/dialogs/applicationpardlg.h
     DSView/pv/dock/keywordlineedit.h

dreamsource-tai, 2022-10-17:
@tysenmoore-xse 
Sorry! Please pull the code again.