DreamSourceLab/DSView #557: Feature request: Maximizing window not fullscreen but Windowed fullscreen

Decee1, 2022-11-10:
Hello,

It would be great if the full screen per default is not enabled and then make it possible to enable it.
I am unsure if this is for all OS's but on Windows normally when you maximize it goes into windowed fullscreen and not fullscreen.
It should only only be in windowed fullscreen when maximizing except if the fullscreen option is enabled. This option should be placed under "Display"

dreamsource-tai, 2022-11-11:
@Decee1 
Because the display scaling parameters of the user system are different, the maximum adaptation is not so satisfactory. You can click the button on the top right to perform a regular display. Then the maximum display will be better.

Decee1, 2022-11-11:
@dreamsource-taiI updated the description. I felt like i wrote it incorrectly. 

The whole thing is. When you maximize it goes to fullscreen 
Maximize != fullscreen