DreamSourceLab/DSView #559: S key binds to "Start", including if CTRL/COMMAND/SHIFT are held down

joshterrill, 2022-11-13:
Right now, pressing the S key in any context (like S, COMMAND+S, COMMAND+SHIFT+S, CTRL+S, etc.) all run the "Start" command.

I would definitely expect that COMMAND+S and COMMAND+SHIFT+S would behave just like all other applications behave, and will let you Save/Overwrite Existing Save, and Save New.

Unfortunately, I was working inside a logic capture, and wanted to save my data, so I pressed COMMAND+S and it started a fresh capture and I lost my work. I can't imagine I'm the first person that has done this. I'm also happy to make a pull request for this if the maintainers agree with the change.

dreamsource-tai, 2022-11-14:
@joshterrill 
Ok, we will fix this problem,thank you!

joshterrill, 2022-11-14:
@dreamsource-tai Is the codebase in the current master branch able to be built? I believe I have the changes required in a feature branch and would submit a PR. But I'm having to make several changes to the codebase just to get it to build. I noticed in previous master commits, it looks like there's some major code-refactoring going on. So I switched to the release-1.2.1 branch and was able to build it, but it's missing a bunch of functions that this new keyboard shortcut code needs. Could I just submit a PR and ask you to test it to make sure it works as expected?

dreamsource-tai, 2022-11-15:
@joshterrill 
This problem has been solved and will be released in a version.

joshterrill, 2022-11-15:
@dreamsource-tai great, I'll look forward to the commit and new version.

closed 2022-11-15 by joshterrill