DreamSourceLab/DSView #563: File -> Screenshot crashes DSView in MacOS Ventura 13.0

joshterrill, 2022-11-27:
Was wanting to export a screenshot today, and clicked File -> Screenshot and the program crashed unexpectedly, and I lost all of my capture info. 

Here is the info from the crash report if it helps:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               DSView [21158]
Path:                  /Applications/DSView.app/Contents/MacOS/DSView
Identifier:            com.yourcompany.DSView
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-11-26 23:20:10.0643 -0800
OS Version:            macOS 13.0 (22A380)
Report Version:        12
Bridge OS Version:     7.0 (20P411)
Anonymous UUID:        905538A1-8FFA-B2F4-77D1-4D49400CBC73

Sleep/Wake UUID:       129B51A7-CD40-4E40-BD74-49F46280B27D

Time Awake Since Boot: 370000 seconds
Time Since Wake:       10207 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 8 Floating point exception: 8
Terminating Process:   exc handler [21158]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libqcocoa.dylib               	       0x102f40b1d 0x102f31000 + 64285
1   QtGui                         	       0x10393fe87 QScreen::grabWindow(unsigned long long, int, int, int, int) + 263
2   DSView                        	       0x10219059c pv::MainWindow::on_screenShot() + 380
3   QtCore                        	       0x1046491d1 QObject::event(QEvent*) + 753
4   QtWidgets                     	       0x1033736da QWidget::event(QEvent*) + 4746
5   QtWidgets                     	       0x103483bf5 QMainWindow::event(QEvent*) + 277
6   QtWidgets                     	       0x10333676d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
7   QtWidgets                     	       0x103337b72 QApplication::notify(QObject*, QEvent*) + 594
8   QtCore                        	       0x10461f7c4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
9   QtCore                        	       0x1046209fe QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 878
10  libqcocoa.dylib               	       0x102f66d69 0x102f31000 + 220521
11  libqcocoa.dylib               	       0x102f674d8 0x102f31000 + 222424
12  CoreFoundation                	    0x7ff800c3fbc8 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
13  CoreFoundation                	    0x7ff800c3fb77 __CFRunLoopDoSource0 + 157
14  CoreFoundation                	    0x7ff800c3f951 __CFRunLoopDoSources0 + 212
15  CoreFoundation                	    0x7ff800c3e5cb __CFRunLoopRun + 929
16  CoreFoundation                	    0x7ff800c3dbb0 CFRunLoopRunSpecific + 560
17  HIToolbox                     	    0x7ff80a51fbd6 RunCurrentEventLoopInMode + 292
18  HIToolbox                     	    0x7ff80a51f9e6 ReceiveNextEventCommon + 679
19  HIToolbox                     	    0x7ff80a51f723 _BlockUntilNextEventMatchingListInModeWithFilter + 70
20  AppKit                        	    0x7ff803d60b37 _DPSNextEvent + 909
21  AppKit                        	    0x7ff803d5f9b8 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1219
22  AppKit                        	    0x7ff803d51ff3 -[NSApplication run] + 586
23  libqcocoa.dylib               	       0x102f66253 0x102f31000 + 217683
24  QtCore                        	       0x10461addf QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
25  QtCore                        	       0x10461fdd2 QCoreApplication::exec() + 130
26  DSView                        	       0x10217d9b1 main + 945
27  dyld                          	    0x7ff800834310 start + 2432

Thread 1:: org.libusb.device-hotplug
0   libsystem_kernel.dylib        	    0x7ff800b266a2 mach_msg2_trap + 10
[... 434 lines cut ...]
Network Service: Wi-Fi, AirPort, en0
USB Device: USB3.0 Hub
USB Device: USB3.1 Hub
USB Device: USB3.1 Hub
USB Device: USB-based DSL Instrument v2
USB Device: USB31Bus
USB Device: N/A
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard
USB Device: PD3.0 USB-C Device
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

```

dreamsource-tai, 2022-11-28:
@joshterrill 
What is your DSView version?

joshterrill, 2022-11-28:
DSView 1.2.1 (x64)

dreamsource-tai, 2022-11-30:
@joshterrill 
Does it break down every time? How many monitors do you have?

joshterrill, 2022-11-30:
A couple of things that I noticed. I do have multiple monitors, and when I disconnect all external monitors, the capture does "work", however, all it does it take a full screenshot of my monitor, with the DSView toolbars and sidemenu's, my dock, my other windows, my mouse, etc. everything in view. I would expect that if I was inside an app and went to click a button that says "Capture", that it would capture only what is in view of that app. I wouldn't expect it to take a picture of the app's own toolbars and sidemenu's, let alone the rest of my screen including dock, etc.