DreamSourceLab/DSView #567: Cannot switch to Oscilloscope mode on original DSLogic (works on 0.99)

piit79, 2022-12-02:
Hello,

I have the original DSLogic Full Kit from Kicstarter (Core, Oscilloscope, Logic Analyser, wireless acquisition board etc.). I was able to use all three modes (Oscilloscope, Logic Analyser, Data Acquisition) on 0.99 (obtained from https://launchpad.net/~tsaarni/+archive/ubuntu/dreamsourcelab-dsview and installed on my one my older machines).

However, only Logic Analyser mode works in recent DSView versions - there is no Oscilloscope or Data Acquisition option in the mode dropdown menu.

I found that the capabilities of the DSLogic device in libsgrok4DSL were changed in v1.0.0 (commit 45204e6a8db540ebd89b7fead43fabeec028ec1a - the whole file [libsigrok4DSL/hardware/DSL/dsl.h](https://github.com/DreamSourceLab/DSView/blob/45204e6a8db540ebd89b7fead43fabeec028ec1a/libsigrok4DSL/hardware/DSL/dsl.h) was replaced; capabilities line in the [previous commit](https://github.com/DreamSourceLab/DSView/blob/856e1803a7883e5d5e4421d476a8a2a8193ec7f9/libsigrok4DSL/hardware/DSL/dsl.h#L281)) - the capabilities `CAPS_MODE_LOGIC | CAPS_MODE_ANALOG | CAPS_MODE_DSO` were replaced with just `CAPS_MODE_LOGIC`.

Why would that be? Would it be possible to get the Oscilloscope and Data Acquisition modes back please? I'd love to use my device to its full potential.

When I add the capabilities to `dsl.h` and recompile, the two modes appear in the menu, but they don't work - no waveforms can be captured, the channel offset cannot be set (the marker is missing), the trigger level cannot be set (the marker cannot be dragged) etc.

Here's the log from the application:

```
sr: dslogic: config_set: setting mode to 1
sr: dslogic: config_set: setting Signal Max Height to 0
sr: dslogic: config_set: setting pattern to 1
sr: dslogic: config_set: setting channel mode to 28
sr: dslogic: config_set: setting filter to 0
sr: dslogic: config_set: setting DSO Trigger Slope to 0
sr: dslogic: config_set: setting DSO Trigger Source to 1
sr: dslogic: config_set: setting DSO Horiz Trigger Position to 10000
sr: dslogic: config_set: setting Trigger Holdoff Time to 0
sr: dslogic: config_set: setting Trigger Margin to 8
sr: dslogic: config_set: setting OFFSET of channel 0 to 0
sr: dslogic: config_set: setting channel 0 Trigger Value to 240
sr: dslogic: config_set: setting ENABLE of channel 0 to 1
sr: dslogic: config_set: setting VDIV of channel 0 to 500 mv
sr: dslogic: config_set: setting Factor of channel 0 to 10
sr: dslogic: config_set: setting AC COUPLING of channel 0 to 1
sr: dslogic: config_set: setting OFFSET of channel 0 to 0
sr: dslogic: config_set: setting channel 0 Trigger Value to 240
sr: dslogic: config_set: setting OFFSET of channel 1 to 0
sr: dslogic: config_set: setting channel 1 Trigger Value to 240
sr: dslogic: config_set: setting ENABLE of channel 1 to 1
sr: dslogic: config_set: setting VDIV of channel 1 to 1000 mv
sr: dslogic: config_set: setting Factor of channel 1 to 1
sr: dslogic: config_set: setting AC COUPLING of channel 1 to 0
sr: dslogic: config_set: setting OFFSET of channel 1 to 0
sr: dslogic: config_set: setting channel 1 Trigger Value to 240
DSView: StoreSession::load_decoders(), is not LOGIC mode.
sr: dslogic: config_set: setting channel 0 Trigger Value to 240
... repeated many times ...
sr: dslogic: config_set: setting ENABLE of channel 1 to 0
sr: dslogic: config_set: setting ENABLE of channel 1 to 1
sr: dslogic: config_set: setting ENABLE of channel 1 to 0
sr: dslogic: config_set: setting AC COUPLING of channel 0 to 0
sr: dslogic: config_set: setting AC COUPLING of channel 0 to 1
sr: dslogic: config_set: setting AC COUPLING of channel 0 to 0
sr: dslogic: config_set: setting AC COUPLING of channel 0 to 1
sr: dslogic: config_set: setting Factor of channel 0 to 1
sr: dslogic: config_set: setting Factor of channel 0 to 10
sr: dslogic: config_set: setting OFFSET of channel 0 to 0
sr: dslogic: config_set: setting channel 0 Trigger Value to 240
sr: dslogic: config_set: setting OFFSET of channel 0 to 0
... repeated many times ...
sr: dslogic: config_set: setting OFFSET of channel 0 to 0
sr: dslogic: config_set: setting OFFSET of channel 1 to 0
sr: session: Starting...
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
sr: Arm FPGA done
sr: dslogic: Starting acquisition.
sr: dslogic: Sending SR_DF_HEADER packet.
DSView: session run loop start
sr: session: Running...
sr: dslogic: config_set: setting OFFSET of channel 0 to 0
sr: dslogic: config_set: setting channel 0 Trigger Value to 240
... repeated ~15 times...
sr: Unable to send CMD_CTL_RD command: LIBUSB_ERROR_OVERFLOW.
sr: dslogic: Failed to get progress infos.
```

I tried building at least v0.99, but it doesn't even compile with recent versions of gcc due to some STL templating issues (gcc 10-12; gcc 9 is broken, and gcc 8 is not available in my Ubuntu version install any more).

Is there anything I can do to help?

Thanks a lot!
Pete

dreamsource-tai, 2022-12-05:
@piit79 
Your device can works with v0.99 only, if you clone the source code from gihub, please reset the commit to v0.99.
Can you upload the compile log?

DreamSourceLab, 2022-12-08:
@piit79 
Sorry for any inconvenience.
After DSView v1.0.0, there are no FPGA resources to support logic analyzer and oscilloscope functions meanwhile.
And we also provide an extra coupon ($120) for kickstarter pledges.
So they can upgrade the oscilloscope addon to standalone oscilloscope [DSCope](https://www.dreamsourcelab.com/shop-category/oscilloscope/).

piit79, 2022-12-08:
Do I understand correctly the FPGA is not powerful enough to support the Oscilloscope function at all starting from v1.0.0?

Thanks for mentioning the promotion for Kickstarter backers. I would most probably be interested and I contacted sales.

DreamSourceLab, 2022-12-09:
@piit79 
The FPGA is not powerful enough to support the Oscilloscope and Logic analyzer function at the same time from v1.0.0.
Thank you for your understanding.

piit79, 2022-12-09:
Ah, I see - "at the same time"... Wouldn't it be easy enough to have two versions of the firmware, one for logic analyzer and one for oscilloscope that people could swap between and just unplug/replug the DSLogic?

But I get you're probably not very keen to invest time in supporting an old device and you'd rather sell your new ones. Still, I wish a better solution was found than simply dropping the support at v1.0.0...

Thanks for your fast responses anyway, I appreciate it.