DreamSourceLab/DSView #570: DSView v0.9.9 build fails with recent gcc versions (10-12)

piit79, 2022-12-06:
Because Oscilloscope/DAQ mode support for the original DSLogic from the Kicstarter was apparently (and very sadly) dropped in v1.0.0 (related issue: #567), I tried building v0.9.9 in Ubuntu 22.10 (Kinetic). However, it fails with all the gcc versions available (9-12).

### gcc 10-12

1) First, the compilation of DSView fails with the following message:
```
/home/error/src/DSView-old/DSView/pv/view/viewport.cpp:1385:38: error: variable ‘QPainterPath path’ has initializer but incomplete type
 1385 |             QPainterPath path(QPoint(_cur_preX, _cur_preY));
```
This is fixed easily by adding
```
#include <QPainterPath>
```
to `DSView/pv/view/viewport.cpp`.

2) Then, the compilation fails with the following error:
```
[ 17%] Building CXX object CMakeFiles/DSView.dir/pv/data/decode/decoder.cpp.o
In file included from /usr/include/c++/12/bits/stl_algobase.h:64,
                 from /usr/include/c++/12/bits/stl_tree.h:63,
                 from /usr/include/c++/12/map:60,
                 from /home/error/src/DSView-old/DSView/pv/data/decode/decoder.h:25,
                 from /home/error/src/DSView-old/DSView/pv/data/decode/decoder.cpp:24:
/usr/include/c++/12/bits/stl_pair.h: In instantiation of ‘struct std::pair<const std::__cxx11::basic_string<char>, _GVariant*>’:
/home/error/src/DSView-old/DSView/pv/data/decode/decoder.cpp:45:32:   required from here
/usr/include/c++/12/bits/stl_pair.h:191:11: error: ‘std::pair<_T1, _T2>::first’ has incomplete type
  191 |       _T1 first;                 ///< The first member
      |           ^~~~~
```
Unfortunately I wasn't able to figure out how to fix this.

### gcc 8

I was able to build DSView v0.9.9 successfully on Ubuntu 18.04 (Bionic) with gcc-8 (in a Docker container).

### gcc 9

The build failedin Ubuntu 20.04 (Focal) with the default gcc-9 at the very end:

```
[100%] Linking CXX executable DSView
/usr/bin/cmake -E cmake_link_script CMakeFiles/DSView.dir/link.txt --verbose=1
/usr/bin/c++    -O2 -g -DNDEBUG  -rdynamic CMakeFiles/DSView.dir/main.cpp.o CMakeFiles/DSView.dir/pv/sigsession.cpp.o CMakeFiles/DSView.dir/pv/mainwindow.cpp.o CMakeFiles/DSView.dir/pv/devicemanager.cpp.o CMakeFiles/DSView.dir/pv/data/snapshot.cpp.o CMakeFiles/DSView.dir/pv/data/signaldata.cpp.o CMakeFiles/DSView.dir/pv/data/logicsnapshot.cpp.o CMakeFiles/DSView.dir/pv/data/logic.cpp.o CMakeFiles/DSView.dir/pv/data/analogsnapshot.cpp.o CMakeFiles/DSView.dir/pv/data/analog.cpp.o CMakeFiles/DSView.dir/pv/dialogs/deviceoptions.cpp.o CMakeFiles/DSView.dir/pv/prop/property.cpp.o CMakeFiles/DSView.dir/pv/prop/int.cpp.o CMakeFiles/DSView.dir/pv/prop/enum.cpp.o CMakeFiles/DSView.dir/pv/prop/double.cpp.o CMakeFiles/DSView.dir/pv/prop/bool.cpp.o CMakeFiles/DSView.dir/pv/prop/binding/binding.cpp.o CMakeFiles/DSView.dir/pv/toolbars/samplingbar.cpp.o CMakeFiles/DSView.dir/pv/view/viewport.cpp.o CMakeFiles/DSView.dir/pv/view/view.cpp.o CMakeFiles/DSView.dir/pv/view/timemarker.cpp.o CMakeFiles/DSView.dir/pv/view/signal.cpp.o CMakeFiles/DSView.dir/pv/view/ruler.cpp.o CMakeFiles/DSView.dir/pv/view/logicsignal.cpp.o CMakeFiles/DSView.dir/pv/view/header.cpp.o CMakeFiles/DSView.dir/pv/view/cursor.cpp.o CMakeFiles/DSView.dir/pv/view/analogsignal.cpp.o CMakeFiles/DSView.dir/pv/prop/binding/deviceoptions.cpp.o CMakeFiles/DSView.dir/pv/toolbars/trigbar.cpp.o CMakeFiles/DSView.dir/pv/toolbars/filebar.cpp.o CMakeFiles/DSView.dir/pv/dock/protocoldock.cpp.o CMakeFiles/DSView.dir/pv/dock/triggerdock.cpp.o CMakeFiles/DSView.dir/pv/dock/measuredock.cpp.o CMakeFiles/DSView.dir/pv/dock/searchdock.cpp.o CMakeFiles/DSView.dir/pv/toolbars/logobar.cpp.o CMakeFiles/DSView.dir/pv/data/groupsnapshot.cpp.o CMakeFiles/DSView.dir/pv/view/groupsignal.cpp.o CMakeFiles/DSView.dir/pv/data/group.cpp.o CMakeFiles/DSView.dir/pv/dialogs/about.cpp.o CMakeFiles/DSView.dir/pv/dialogs/search.cpp.o CMakeFiles/DSView.dir/pv/data/dsosnapshot.cpp.o CMakeFiles/DSView.dir/pv/data/dso.cpp.o CMakeFiles/DSView.dir/pv/view/dsosignal.cpp.o CMakeFiles/DSView.dir/pv/view/dsldial.cpp.o CMakeFiles/DSView.dir/pv/dock/dsotriggerdock.cpp.o CMakeFiles/DSView.dir/pv/view/trace.cpp.o CMakeFiles/DSView.dir/pv/view/selectableitem.cpp.o CMakeFiles/DSView.dir/pv/data/decoderstack.cpp.o CMakeFiles/DSView.dir/pv/data/decode/rowdata.cpp.o CMakeFiles/DSView.dir/pv/data/decode/row.cpp.o CMakeFiles/DSView.dir/pv/data/decode/decoder.cpp.o CMakeFiles/DSView.dir/pv/data/decode/annotation.cpp.o CMakeFiles/DSView.dir/pv/view/decodetrace.cpp.o CMakeFiles/DSView.dir/pv/prop/binding/decoderoptions.cpp.o CMakeFiles/DSView.dir/pv/widgets/fakelineedit.cpp.o CMakeFiles/DSView.dir/pv/widgets/decodermenu.cpp.o CMakeFiles/DSView.dir/pv/widgets/decodergroupbox.cpp.o CMakeFiles/DSView.dir/pv/prop/string.cpp.o CMakeFiles/DSView.dir/pv/device/sessionfile.cpp.o CMakeFiles/DSView.dir/pv/device/inputfile.cpp.o CMakeFiles/DSView.dir/pv/device/file.cpp.o CMakeFiles/DSView.dir/pv/device/devinst.cpp.o CMakeFiles/DSView.dir/pv/dialogs/storeprogress.cpp.o CMakeFiles/DSView.dir/pv/storesession.cpp.o CMakeFiles/DSView.dir/pv/view/devmode.cpp.o CMakeFiles/DSView.dir/pv/device/device.cpp.o CMakeFiles/DSView.dir/pv/dialogs/waitingdialog.cpp.o CMakeFiles/DSView.dir/pv/dialogs/dsomeasure.cpp.o CMakeFiles/DSView.dir/pv/dialogs/calibration.cpp.o CMakeFiles/DSView.dir/pv/data/decodermodel.cpp.o CMakeFiles/DSView.dir/pv/dialogs/protocollist.cpp.o CMakeFiles/DSView.dir/pv/dialogs/protocolexp.cpp.o CMakeFiles/DSView.dir/pv/dialogs/fftoptions.cpp.o CMakeFiles/DSView.dir/pv/data/mathstack.cpp.o CMakeFiles/DSView.dir/pv/view/mathtrace.cpp.o CMakeFiles/DSView.dir/dsapplication.cpp.o CMakeFiles/DSView.dir/pv/widgets/viewstatus.cpp.o CMakeFiles/DSView.dir/pv/toolbars/titlebar.cpp.o CMakeFiles/DSView.dir/pv/mainframe.cpp.o CMakeFiles/DSView.dir/pv/widgets/border.cpp.o CMakeFiles/DSView.dir/pv/dialogs/dsmessagebox.cpp.o CMakeFiles/DSView.dir/pv/dialogs/shadow.cpp.o CMakeFiles/DSView.dir/pv/dialogs/dsdialog.cpp.o CMakeFiles/DSView.dir/pv/dialogs/interval.cpp.o CMakeFiles/DSView.dir/pv/prop/binding/probeoptions.cpp.o CMakeFiles/DSView.dir/pv/moc_sigsession.cpp.o CMakeFiles/DSView.dir/pv/moc_mainwindow.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_deviceoptions.cpp.o CMakeFiles/DSView.dir/pv/prop/moc_property.cpp.o CMakeFiles/DSView.dir/pv/prop/moc_int.cpp.o CMakeFiles/DSView.dir/pv/prop/moc_enum.cpp.o CMakeFiles/DSView.dir/pv/prop/moc_double.cpp.o CMakeFiles/DSView.dir/pv/prop/moc_bool.cpp.o CMakeFiles/DSView.dir/pv/toolbars/moc_samplingbar.cpp.o CMakeFiles/DSView.dir/pv/view/moc_viewport.cpp.o CMakeFiles/DSView.dir/pv/view/moc_view.cpp.o CMakeFiles/DSView.dir/pv/view/moc_timemarker.cpp.o CMakeFiles/DSView.dir/pv/view/moc_ruler.cpp.o CMakeFiles/DSView.dir/pv/view/moc_header.cpp.o CMakeFiles/DSView.dir/pv/view/moc_cursor.cpp.o CMakeFiles/DSView.dir/pv/toolbars/moc_trigbar.cpp.o CMakeFiles/DSView.dir/pv/toolbars/moc_filebar.cpp.o CMakeFiles/DSView.dir/pv/dock/moc_protocoldock.cpp.o CMakeFiles/DSView.dir/pv/dock/moc_triggerdock.cpp.o CMakeFiles/DSView.dir/pv/dock/moc_measuredock.cpp.o CMakeFiles/DSView.dir/pv/dock/moc_searchdock.cpp.o CMakeFiles/DSView.dir/pv/toolbars/moc_logobar.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_about.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_search.cpp.o CMakeFiles/DSView.dir/pv/dock/moc_dsotriggerdock.cpp.o CMakeFiles/DSView.dir/pv/view/moc_trace.cpp.o CMakeFiles/DSView.dir/pv/view/moc_selectableitem.cpp.o CMakeFiles/DSView.dir/pv/data/moc_decoderstack.cpp.o CMakeFiles/DSView.dir/pv/view/moc_decodetrace.cpp.o CMakeFiles/DSView.dir/pv/widgets/moc_fakelineedit.cpp.o CMakeFiles/DSView.dir/pv/widgets/moc_decodermenu.cpp.o CMakeFiles/DSView.dir/pv/widgets/moc_decodergroupbox.cpp.o CMakeFiles/DSView.dir/pv/prop/moc_string.cpp.o CMakeFiles/DSView.dir/pv/device/moc_devinst.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_storeprogress.cpp.o CMakeFiles/DSView.dir/pv/moc_storesession.cpp.o CMakeFiles/DSView.dir/pv/view/moc_devmode.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_waitingdialog.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_dsomeasure.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_calibration.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_protocollist.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_protocolexp.cpp.o CMakeFiles/DSView.dir/pv/dialogs/moc_fftoptions.cpp.o CMakeFiles/DSView.dir/pv/data/moc_mathstack.cpp.o CMakeFiles/DSView.dir/pv/view/moc_mathtrace.cpp.o CMakeFiles/DSView.dir/pv/widgets/moc_viewstatus.cpp.o CMakeFiles/DSView.dir/pv/toolbars/moc_titlebar.cpp.o CMakeFiles/DSView.dir/pv/moc_mainframe.cpp.o CMakeFiles/DSView.dir/pv/widgets/moc_bo [... cut ...]
/usr/bin/ld: /usr/local/lib/libsigrokdecode4DSL.so: undefined reference to `PyModule_AddIntConstant'
/usr/bin/ld: /usr/local/lib/libsigrokdecode4DSL.so: undefined reference to `PyList_New'
/usr/bin/ld: /usr/local/lib/libsigrokdecode4DSL.so: undefined reference to `PyObject_Str'
/usr/bin/ld: /usr/local/lib/libsigrokdecode4DSL.so: undefined reference to `PyErr_Format'
/usr/bin/ld: /usr/local/lib/libsigrokdecode4DSL.so: undefined reference to `PyTuple_Size'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/DSView.dir/build.make:2506: DSView] Error 1
```

dreamsource-tai, 2022-12-07:
@piit79 
Maybe your compilation script doesn't handle the Python library well. Please refer to the latest code compilation script, and then modify the v0.99 compilation script to try. It is an error in the linking process, not the cause of gcc.

piit79, 2022-12-07:
Well, it's not "my compilation script", it's the same compilation script that works well in the older Ubuntu version with gcc-8 :) Anyway, I'll see if I can find any differences in the latest code.

However, you only addressed the last point of the issue. Do you have any idea what's going on with the other errors in gcc 10-12?

```
[ 17%] Building CXX object CMakeFiles/DSView.dir/pv/data/decode/decoder.cpp.o
In file included from /usr/include/c++/12/bits/stl_algobase.h:64,
                 from /usr/include/c++/12/bits/stl_tree.h:63,
                 from /usr/include/c++/12/map:60,
                 from /home/error/src/DSView-old/DSView/pv/data/decode/decoder.h:25,
                 from /home/error/src/DSView-old/DSView/pv/data/decode/decoder.cpp:24:
/usr/include/c++/12/bits/stl_pair.h: In instantiation of ‘struct std::pair<const std::__cxx11::basic_string<char>, _GVariant*>’:
/home/error/src/DSView-old/DSView/pv/data/decode/decoder.cpp:45:32:   required from here
/usr/include/c++/12/bits/stl_pair.h:191:11: error: ‘std::pair<_T1, _T2>::first’ has incomplete type
  191 |       _T1 first;                 ///< The first member
      |           ^~~~~
```

I'd love to build DSView natively and not run it from a docker container...

It's such a shame oscilloscope support for the original DSLogic was dropped in v1.0.0, otherwise I could just run the latest version of DSView that builds fine...

DreamSourceLab, 2022-12-08:
@piit79 
Sorry for any inconvenience.
After DSView v1.0.0, there are no FPGA resources to support logic analyzer and oscilloscope functions meanwhile.
And we also provide an extra coupon ($120) for kickstarter pledges.
So they can upgrade the oscilloscope addon to standalone oscilloscope [DSCope](https://www.dreamsourcelab.com/shop-category/oscilloscope/).