DreamSourceLab/DSView #580: I2C protocol decoder doesn't decode protocol correctly.

r12f, 2022-12-23:
I2C protocol decoder doesn't follow clock line correctly and results in the data line being decoded incorrectly.

Here is the graph showing the result:

[image]

DreamSourceLab, 2022-12-25:
For the I2C signal, you set a sample rate that is too high (500MHz). The slow transition edge of I2C signal is sampled multiple times, resulting in multiple transitions being acquired along one edge. If you zoom in on the waveform, you will notice that multiple edges are shrunk together. We generally recommend a sampling rate of 10x-100x

r12f, 2022-12-26:
hi @DreamSourceLab , thank you so much for helping! I see! I have lowered the sample rate to 10Mhz and it works like a charm! All signals are parsed correctly now! 

Thanks again! and I have marked this issue as closed.

closed 2022-12-26 by r12f