DreamSourceLab/DSView #585: Advance triggering and filtering capabilities ?

mtk11, 2023-01-07:
Interested if advance triggering and filtering capabilities might be available or planned. For example how to process with a hypothetical scenario described above:

A system with a 100MHz clock, an address bus, a data bus, and an address valid signal. Every 1ms data is read, and you are only interested in reads from address in the range 1234 to 5678. You want to capture (up to) 10000 values read on the rising edge of the clock provided address valid is asserted and provided the address range is right.

dreamsource-tai, 2023-01-13:
@mtk11 
Ok, we have collected your ideas.

rpv-tomsk, 2024-03-04:
That is not hypothetical scenario ) I'm here with a same - need to debug IO with chip on a such bus.
Currently I'm interested to filter chip IO from all bus events based on CE# signal (simpler scenario than yours)

bkw777, 2024-09-03:
I want to be able to decode parallel based on multiple conditions, not just a single clk line. /CE low plus ALE falling plus /OE maybe high maybe low at different times.

I'd really like to also be able to add any arbitrary conditions so I can trap when something is accessed by an address, ie a particular set of data and/or address lines. Like the search function where you can set XFR10 for all lines, but in the parallel decoder in place of the CLK field.

The bus has continuous traffic but I need to see just one chip, and /CE isn't good enough by itself because within the window while /CE is low, several different things happen thanks to ALE, /OE, /WE. For an example: The data at the exact time of /CE is not meaningful, the data at the time of ALE falling is meaningful, but only after /CE is already low.