DreamSourceLab/DSView #587: 【新特性】是否可以在界面底部增加一个界面，用于显示解码后的信息

oxlm, 2023-01-28:
[image]
类似于这种界面，在选中底层的数据时，顶层也可以直接切换到对应位置。

这种显示方式在查看串口数据，红外数据，USB数据，I2C数据等一收一发的数据时，会很直观

dreamsource-tai, 2023-01-28:
@oxlm 
DSView右边数据列表里的数据，单击后，可以切换到对应的波形段。不知道你想要的是什么样的功能？不太理解你的意思。

oxlm, 2023-02-01:
大概是这么个意思，解码后不是简单的按照字节的方式显示，而是更详细一点，串口是否可以显示ascii内容表？usb、I2C、SPI是否可以按帧去分帧？红外是否有码值和repeat帧的合并显示？

oxlm, 2023-02-01:
主要还是下面那个标红的部分的显示，如果能做到分帧显示，并且数据也能按照类似的方式导出，会大大降低后续数据分析的难度

dreamsource-tai, 2023-02-02:
@oxlm 
哪些数据是一帧的，只有协议自己做出分析才能识别。DSView那么多协议，所有协议面对的都是一样的数据流，它们不可能识别到一帧数据有哪些，然后做出像你这样的按帧分组。不过可以通过修改协议，在帧的开始和结束位置输出标记文字，比如start和end，帮助人们区分帧的范围。

CloverGit, 2023-05-16:
@dreamsource-tai 我觉得他的意思应该是希望能添加一个水平显示数据列表的功能，DSView目前只有右边垂直的数据列表显示。这样按照时间来线性排列数据列表确实更加直观。