DreamSourceLab/DSView #590: Doesn't work after restarting PC

juf00th, 2023-02-08:
If the u3p100 is not disconnected from the PC, after restarting the PC, the red LED will flash and the device will not work.
How can I do that would not reconnect the u3p100?

os: Linux 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux
dsview 1.2.2
device: u3p100

dreamsource-tai, 2023-02-09:
@juf00th 
Have you reconnected the device?

dreamsource-tai, 2023-02-09:
@juf00th 
Disconnect and reconnect the USB data cable.

juf00th, 2023-02-09:
@dreamsource-tai Of course if you reconnect USB everything works. But when you often use an oscilloscope and it is always connected, this is not convenient. For example, the logic analyzer (dslogic plus) does not have this problem. Maybe somehow you can fix this thing for u3p100? Thank you.

dreamsource-tai, 2023-02-10:
@juf00th 
OK, we will optimize this problem in the future version.

juf00th, 2023-02-10:
@dreamsource-tai Any ideas why this is happening? Maybe I can fix it myself.

dreamsource-tai, 2023-02-13:
@juf00th 
We will try to fix this problem in the new version, and it depends on whether the platform supports it.

mrehkopf, 2023-07-07:
I have the same issue with a DSLogic U3Pro32.
It does not like being detached; I tried to use it with usbipd so I could try a linux build of DSView in WSL (because there are no instructions for Windows builds and I couldn't get it working), but whenever usbipd detaches and attaches to WSL it starts flashing the LED as well and software simply says it's busy.
I think it would need to be fixed in firmware.

juf00th, 2023-07-07:
@mrehkopf 
I could be wrong, but it looks like a hardware problem. The microcircuit responsible for the interface does not return to its initial state after unmount the device.