DreamSourceLab/DSView #592: SPI decoder issue

MrdUkk, 2023-02-23:
i think this is is bug in decoding SPI bus. below is capture. the signals are the same each time. but decoder fail to correctly display individual bits

[image]

used options
[image]

DreamSourceLab, 2023-02-24:
@MrdUkk 
It seems that your CLK signal and MOSI signal do not meet the setup and hold conditions (they just coincide), either you need to set a different CPOL, or choose a higher sample rate to see if you can tell the difference between the phase of CLK and MOSI

MrdUkk, 2023-02-27:
@DreamSourceLab Strange. I will check again.
This is 1Mhz test pattern signal comes by 4cm traces on PCB from AVR uC to 74hc family. This is more than twice of required by Nyquist frequency. Captured at 20mhz as seen in screenshots.
Im using hardware SPI capability presented by uC itself.

we clearly see signals at PINs on lower part of screenshot.
Each of 3 presented data exchanges is completely and cleanly seen.

Why only decoder fails to visualize bits? (upper part of screenshot).

radorodopski, 2023-07-07:
> This is more than twice of required by Nyquist frequency. Captured at 20mhz as seen in screenshots. Im using hardware SPI capability presented by uC itself.
Actually, for analog signals it is 2x times, and for digital signals it is at least 5x or even 10x if you want to capture the actual duty cycle of a clock (in 5x it will be distorted randomly based on when the clock pulses coincide with the sample clock of the logic analyzer). In your case it is 20x, which is fine for digital signals.

However, it seems the CPOL of your host and slave devices mismatches - one device sends data on the rising edge (slave), and the other on the falling edge (master) which shouldn't be expected to work. I think you should use the same CPHA on both devices and set that same CPHA into DSView - and it will properly decode your data. In your case you're trying to sample the data on the same edge it is shifted out, which may produce incorrect results.

In case I may be missing anything, please feel free to correct me.