DreamSourceLab/DSView #595: Per channel threshold level

RussianE39, 2023-03-09:
Is it possible to implement per-channel threshold level? Use case: Recording of both CAN-H and CAN-L signals.

radorodopski, 2023-07-07:
Hi, folks. Any feedback on this?

I don't think it is possible in the current devices (or I may be wrong, as I am not quite familiar with the hardware yet) but maybe in future devices you could make it possible to change the threshold per 8-bit bank maybe (meaning every 8 or so bits having separate configurable threshold levels) ?

Please let us know if anything like that is possible with any of the current hardware out there. :)

DreamSourceLab, 2023-07-09:
@RussianE39 
For CAN bus, you can just record CAN-L singal for decoding

DreamSourceLab, 2023-07-09:
@radorodopski 
Yes, current device don't support mutiple threshold for different channels.
And we will take this into our consideration for future devcies.