DreamSourceLab/DSView #596: IIC data analysis problem,When the scale is reduced, the data content is not displayed

reliXQ, 2023-03-13:
"Start" is too small to show the decoded data when zoomed to Figure 2
Please look at the picture below to explain, thanks!
[image]

reliXQ, 2023-03-13:
This state is the best, after all, mainly depends on the parsed content
[image]

reliXQ, 2023-03-13:
This is the original file in question
[DSLogic-la-230313-105248.zip](https://github.com/DreamSourceLab/DSView/files/10953450/DSLogic-la-230313-105248.zip)

dreamsource-tai, 2023-03-14:
@reliXQ 
你可以通过右边的列表，双击放大某一个结果。

dreamsource-tai, 2023-08-25:
@reliXQ 
请看一下v1.3版本，这个问题应该是修复了。

reliXQ, 2023-10-07:
OK,我回头试下.

closed 2023-10-07 by reliXQ