DreamSourceLab/DSView #598: libusb version + patch on Windows

fbouchoux, 2023-03-24:
I am trying to build DSView on Windows using MSYS2 + MinGW64.
However, depending on the libusb version used (1.0.23, 1.0.24, sigrok version tested), DSView crashes when the capture is started and/or stopped. Also, the USB hotplug detection is not working.

I suspect that DSView uses a patched libusb version (for Windows at least).
Do you have the .patch file to use? Which libusb version should we use?

GengKanghua, 2023-04-01:
An error was reported on my computer, it's about the function "libusb_get_pollfds()" in "dscope.c". It was found in the libusb documentation that this function is not supported on Windows. 
“Because there is no native on Windows, the ability to return externally pollable file descriptors on Windows returns an error.poll()libusb_get_pollfd()”
There is an issue about it in libuse. https://github.com/libusb/libusb/issues/252
But I didn't succeed. I am a beginner.

WBF001, 2023-05-25:
I have encountered the same problem. Have you found a solution

andrei7c4, 2024-02-03:
It's pity that @DreamSourceLab is not sharing all the code needed to build functional version of DSView for Windows. They either have patched libusb to get libusb_get_pollfds function working or they avoid use of this function. In either case the changes are not public.