DreamSourceLab/DSView #599: Trackpad unusable for scrolling and zooming (MacOS)

hmoazed, 2023-04-08:
Using the trackpad to scroll through the captured data inadvertently causes zooming in and out, and vise-versa.
This is because the trackpad gesture for scrolling horizontally (two finger left or right swipe) also as a side effect triggers the gesture to zoom in and and out (two finger up or down swipe, which on MacOS trackpad is *not* zooming but moving up and down, zooming is accomplished using the pinch gesture).
To correct this, we'd need to add  "Trackpad / Mouse" toggle, and when in Trackpad mode, the pinch gesture is used to zoom instead of the up/down gesture.
This is a very common issue in cross platform CAD programs (EasyEDA just fixed this recently, as well as Fusion 360).

ddavidebor, 2023-05-06:
Encountered the same issue, very annoying!

dreamsource-tai, 2023-07-03:
@hmoazed 
@ddavidebor 
Ok,We will check this issue. Thank you!

dreamsource-tai, 2023-07-04:
@hmoazed 
@ddavidebor 
We fixed this issue, you can try the new version by the way compile from code source.

M4tsuri, 2024-01-23:
This issue still exists on dev-1.3.2 288a459c93971aea745610c6f3a7553a2b1a1db6

ddavidebor, 2024-01-30:
Hi, sorry, but this took too long, we will buy from Saleae in the future

FearlessArmada, 2025-01-01:
I found a hacky fix, in the file DSView/pv/view/viewport.cpp at line 1360 you have to divide the values by something like 16 or 32, like this:

```
double divisor_zoom = 16; // factor by which it divides
double zoom_scale = delta / 80 / divisor_zoom;

if (ABS_VAL(delta) <= 80){
    zoom_scale = delta > 0 ? 1.5 / divisor_zoom : -1.5 / divisor_zoom;
}
```
This should probably be made into some kind of setting that users can adjust.