DreamSourceLab/DSView #600: Feature request:  Show configuration pathname and filename in DSView title bar

davidk408, 2023-04-25:
It would be helpful to display the path and filename of the currently loaded Configuration file in the title bar of the DSView application.  An asterisk '*' could be displayed at the end of the filename if the configuration has been modified but not saved yet.  

This feature would be especially helpful for those of us that jump between several configuration files.

This feature could be a user-selectable option (e.g. "Display config name in title bar") in the Display>Options menu.

dreamsource-tai, 2023-05-16:
@davidk408 
Ok, We will add this feature at the new version.

dreamsource-tai, 2023-05-16:
@davidk408 
DSView can auto save the config to file after the device detached, and restore the config upon program startup.

dreamsource-tai, 2023-07-03:
@davidk408 
We have added this feature to DSView.
If your operating system is Linux, you can experience the latest features of DSView.