DreamSourceLab/DSView #601: Allow DSView app window height to be resized shorter

davidk408, 2023-04-25:
I often use four or less channels in DSView, so it would be helpful if the height of DSView's main screen could be resized even shorter than it currently allows.  Then we could fit more apps (debuggers, source code editors, etc.) on my screens at the same time,

The attached image shows the shortest resizing I'm able to achieve on my Windows PC (I haven't tested other OSs).  Notice the wasted space at the bottom of the DSView screen.

[image]

dreamsource-tai, 2023-04-26:
@davidk408 
You can click the button in the upper right corner to display the DSView as normal, and then customize the height.

davidk408, 2023-04-26:
@dreamsource-tai 
Thanks for the quick reply.  I am customizing the app height by dragging the lower right corner of the DSView window, but it doesn't allow the window to be very short.  

**PulseView** lets me set the height very short (see attached image below).  In DSView, there must be a "minimum Y size" setting for the DSView window that's preventing me from setting the height any shorter.

[image]

dreamsource-tai, 2023-04-27:
@davidk408 
Ok, we recorded this issue. Thank you!

dreamsource-tai, 2023-07-03:
@davidk408 
We have added this feature to DSView.
If your operating system is Linux, you can experience the latest features of DSView.