DreamSourceLab/DSView #602: xy2-100协议解码界面有空缺

Ging-H, 2023-04-25:
[image]
xy2-100协议解码结果,在某些地方会有空缺,

win10, v1.2.2

dreamsource-tai, 2023-04-26:
@Ging-H 
不理解你的意思，能说具体点吗？

dreamsource-tai, 2023-04-26:
@Ging-H 
是解码结果不对吗？

Ging-H, 2023-04-26:
就是xy2-100的解码选项positions,在某些情况下会不显示奇偶校验错误的数据,
这一段实际上是奇偶校验错误,一般情况下会显示positions解码的数据,并且提示奇偶校验错误
下面是正常显示的奇偶校验错误的解码结果
[image]

dreamsource-tai, 2023-04-27:
@Ging-H 
你能提供数据给我们分析吗？就是保存后的文件，改个名后上传。其实我想知道，这个协议的解码结果是否正确？还是说它是正确的，但是需要完善，让它更好用？

Ging-H, 2023-04-27:
协议功能已经调试好了,现在也没办法重新复现这个问题,
但是观察每一位数据可以看出来那一段数据实际是错误的, 并不是奇偶校验错误,
协议解码结果应该是正确的, 至少能识别出错误的数据, 但是奇偶校验位的显示还是有点问题的,在这种情况下也应该显示奇偶校验'OK'才对的

dreamsource-tai, 2023-04-27:
@Ging-H 
麻烦你提供一份你的数据，我们用于分析，以便完善协议。谢谢！分析议协需要有合适的业务数据。