DreamSourceLab/DSView #605: 在debian/testing上构建1.2.2失败

PCX-LK, 2023-05-03:
```
❯ make -j8
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/mainwindow.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dock/searchdock.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dock/measuredock.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/signal.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/logicsignal.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/toolbars/logobar.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/analogsignal.cpp.o
[  2%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dock/triggerdock.cpp.o
In file included from /usr/include/boost/bind.hpp:30,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/toolbars/logobar.cpp:22:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/trace.h:34,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/signal.h:38,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/signal.cpp:25:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/selectableitem.h:39:31: error: invalid use of incomplete type ‘class QObject’
   39 | class SelectableItem : public QObject
      |                               ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qpainter.h:11,
                 from /usr/include/x86_64-linux-gnu/qt6/QtGui/QPainter:1,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/signal.h:29:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
In file included from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/trace.h:35:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/dsldial.h:33:24: error: invalid use of incomplete type ‘class QObject’
   33 | class dslDial : public QObject
      |                        ^~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
In file included from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/trace.h:34,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/signal.h:38,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/logicsignal.h:27,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/logicsignal.cpp:25:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/selectableitem.h:39:31: error: invalid use of incomplete type ‘class QObject’
   39 | class SelectableItem : public QObject
      |                               ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qpainter.h:11,
                 from /usr/include/x86_64-linux-gnu/qt6/QtGui/QPainter:1,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/signal.h:29:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
In file included from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/trace.h:35:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/dsldial.h:33:24: error: invalid use of incomplete type ‘class QObject’
   33 | class dslDial : public QObject
      |                        ^~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
make[2]: *** [CMakeFiles/DSView.dir/build.make:1042：CMakeFiles/DSView.dir/DSView/pv/view/signal.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[2]: *** [CMakeFiles/DSView.dir/build.make:1070：CMakeFiles/DSView.dir/DSView/pv/view/logicsignal.cpp.o] 错误 1
In file included from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/trace.h:34,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/signal.h:38,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/analogsignal.h:26,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/analogsignal.cpp:24:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/selectableitem.h:39:31: error: invalid use of incomplete type ‘class QObject’
   39 | class SelectableItem : public QObject
      |                               ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qpainter.h:11,
                 from /usr/include/x86_64-linux-gnu/qt6/QtGui/QPainter:1,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/signal.h:29:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
In file included from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/trace.h:35:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/view/../view/dsldial.h:33:24: error: invalid use of incomplete type ‘class QObject’
   33 | class dslDial : public QObject
      |                        ^~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp: In member function ‘void pv::MainWindow::on_session_error()’:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp:660:14: warning: unused variable ‘error_pattern’ [-Wunused-variable]
  660 |     uint64_t error_pattern;
      |              ^~~~~~~~~~~~~
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp: In member function ‘bool pv::MainWindow::gen_session_json(QJsonObject&)’:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp:1161:51: error: use of deleted function ‘QVariant::QVariant(T) [with T = char*; typename std::enable_if<disjunction_v<std::is_pointer<_Tp>, std::is_member_pointer<_Tp> >, bool>::type <anonymous> = false]’
 1161 |     sessionVar["Device"] = QJsonValue::fromVariant(sdi->driver->name);
      |                            ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qaction.h:13,
                 from /usr/include/x86_64-linux-gnu/qt6/QtGui/QAction:1,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp:25:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:199:5: note: declared here
  199 |     QVariant(T) = delete;
      |     ^~~~~~~~
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp: In member function ‘virtual void pv::MainWindow::switchLanguage(int)’:
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp:1444:22: warning: ignoring return value of ‘bool QTranslator::load(const QString&, const QString&, const QString&, const QString&)’, declared with attribute ‘nodiscard’ [-Wunused-result]
 1444 |         _qtTrans.load(":/qt_" + QString::number(language));
      |         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/QTranslator:1,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.h:29,
                 from /home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp:50:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qtranslator.h:33:24: note: declared here
   33 |     [[nodiscard]] bool load(const QString & filename,
      |                        ^~~~
/home/pcxlk/Desktop/DSView-1.2.2/DSView/pv/mainwindow.cpp:1446:22: warning: ignoring return value of ‘bool QTranslator::load(const QString&, const QString&, const QString&, const QString&)’, declared with attribute ‘nodiscard’ [-Wunused-result]
 1446 |         _myTrans.load(":/my_" + QString::number(language));
      |         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qtranslator.h:33:24: note: declared here
   33 |     [[nodiscard]] bool load(const QString & filename,
      |                        ^~~~
make[2]: *** [CMakeFiles/DSView.dir/build.make:1112：CMakeFiles/DSView.dir/DSView/pv/view/analogsignal.cpp.o] 错误 1
make[2]: *** [CMakeFiles/DSView.dir/build.make:776：CMakeFiles/DSView.dir/DSView/pv/mainwindow.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:83：CMakeFiles/DSView.dir/all] 错误 2
make: *** [Makefile:156：all] 错误 2
```

dreamsource-tai, 2023-05-06:
@PCX-LK 
请卸载掉qt6，安装qt5，现在的qt5的包名可能是qt5-default、qtbase5-dev中的一个，安装好了试试。

PCX-LK, 2023-05-06:
> @PCX-LK 请卸载掉qt6，安装qt5，现在的qt5的包名可能是qt5-default、qtbase5-dev中的一个，安装好了试试。

构建成功了
[image]

PCX-LK, 2023-05-06:
[image]

closed 2023-05-06 by PCX-LK