DreamSourceLab/DSView #615: WS2812 decoder not decoding the last LED

AndrejValand, 2023-05-13:
I have tried every setting but can not get the software to decode the last led RGB value. It decodes it maybe in 1 of 50 tries...
In the screenshot below an example of three leds where only the first two are decoded:
[image]

dreamsource-tai, 2023-05-15:
@AndrejValand 
Can you upload your dsl file? We need it to debug the decoder.

AndrejValand, 2023-05-15:
No problem, i hope its an easy fix. As i see it now it does not decode correctlly the last RGB massage, it you can capture more repeats the first massages are correct. If change my software to rapidlly change colours and i capture more pockets all the pockets but the last one are decoded right. When i capture only one LED update massage the last LED in the massage is not decoded.
[Test measurements.zip](https://github.com/DreamSourceLab/DSView/files/11474610/Test.measurements.zip)

dreamsource-tai, 2023-05-15:
@AndrejValand 
Please update the decoder from this file.
[WS2812.zip](https://github.com/DreamSourceLab/DSView/files/11476325/WS2812.zip)

AndrejValand, 2023-05-15:
This was a super fast fix! Thanks. Work OK now. 

[image]

closed 2023-05-15 by AndrejValand