DreamSourceLab/DSView #630: Tracing randomly fails on DSLogic U3Pro16/Linux

hansfbaier, 2023-05-28:
Tracing randomly fails on Linux with DSView commit 0af603c132fe87849fc87fe943e8573021a64f94
built from github.
For example, when I press Instant, then it says:
Triggered! -2147483648% Captured,
and the wait indicator keeps on spinning, and the green LED on the device keeps on blinking
without end, until I abort.
That happens very frequently so that I have to try several times until I get a successful capture.
How can it be solved?

[image]

Ouput of DSView:
```
DSView: ----------------- version: 1.3.0-dev2-----------------
DSView: Qt:5.15.3
sr: lib_main: Firmware resource directory:"/usr/local/share/DSView/res/"
sr: lib_main: Init libsigrok.
sr: demo: Scan demo device.
sr: backend: Register hotplug callback.
sr: lib_main: One device attached,handle:0x5577dae57830
sr: lib_main: Hotplug thread start!
sr: lib_main: Process device attach event.
sr: dslogic: Scan DSLogic device.
sr: dslogic: Found a new device,handle:0x5577dae57830,bus:6,address:2
sr: dslogic: Found a DSLogic device,name:"DSLogic U3Pro16",handle:0x5577dae57830
sr: lib_main: Get new device list by driver "DSLogic"
sr: dscope: Scan DSCope device.
DSView: Decode script files directory:"/usr/local/share/libsigrokdecode4DSL/decoders"
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: device name: "DSLogic U3Pro16".
sr: dsl: dsl_dev_open: Firmware upload was not needed.
sr: dsl: Open usb device instance, handle: 0x5577dae57830
sr: dsl: Opened device 0x5577dae57830 on 6.2, interface 0, firmware 2.1.
sr: dsl: Configure FPGA using "/usr/local/share/DSView/res/DSLogicU3Pro16.bin"
sr: dsl: FPGA configure done: 465028 bytes.
sr: lib_main: Activating device end.
DSView: Switch to device "DSLogic U3Pro16" done.
DSView: Load session file: "/home/jack/.local/share/DreamSourceLab/DSView/DSLogic0.ses.dsc"
DSView: Update rate list.
DSView: Update sample count list.
DSView: Create new decoder,name:"0:SPI",id:"0:spi"
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: Start collect.
sr: lib_main: Start collect.
sr: dsl: dsl_dev_open: Firmware upload was not needed.
sr: lib_main: Collect thread start.
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
sr: dsl: Arm FPGA done
DSView: Stop collect.
DSView: Device is uploading.
sr: dsl: 2582663088: receive_trigger_pos(): status 0; timeout 0; received 1024 bytes.
sr: dsl: finish_acquisition: send SR_DF_END packet
sr: dslogic: remove_sources: remove fds from polling
DSView: ------->decode thread start
srd: Creating session 1.
DSView: decoder sample count: 250000384
sr: lib_main: Collect thread end.
srd: Calling start() of all instances in session 1.
DSView: Start collect.
DSView: send to decoder times: 41
srd: Destroyed session 1.
DSView: ------->decode thread end
sr: lib_main: Start collect.
sr: dsl: dsl_dev_open: Firmware upload was not needed.
sr: lib_main: Collect thread start.
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
sr: dsl: Arm FPGA done
```

dreamsource-tai, 2023-05-29:
@hansfbaier 
You can try the version dev1.3-RC6 of the branch dev-1.3.

hansfbaier, 2023-05-29:
That works better, thanks.
Still get the error sometimes, but much less frequently.

closed 2023-05-29 by hansfbaier