DreamSourceLab/DSView #634: Unable to build under Ubuntu 23.04

priv, 2023-05-31:
Both master and dev-1.3 do not work.

```
/home/alvin/pool/DSView/DSView/pv/view/selectableitem.h:39:31: error: invalid use of incomplete type ‘class QObject’
   39 | class SelectableItem : public QObject
      |                               ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qpainter.h:11,
                 from /usr/include/x86_64-linux-gnu/qt6/QtGui/QPainter:1,
                 from /home/alvin/pool/DSView/DSView/pv/view/signal.h:29:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
In file included from /home/alvin/pool/DSView/DSView/pv/view/trace.h:35:
/home/alvin/pool/DSView/DSView/pv/view/dsldial.h:33:24: error: invalid use of incomplete type ‘class QObject’
   33 | class dslDial : public QObject
      |                        ^~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
/home/alvin/pool/DSView/DSView/pv/view/view.cpp: In member function ‘virtual bool pv::view::View::eventFilter(QObject*, QEvent*)’:
/home/alvin/pool/DSView/DSView/pv/view/view.cpp:722:64: warning: ‘int QMouseEvent::y() const’ is deprecated: Use position() [-Wdeprecated-declarations]
  722 |                         _hover_point = QPoint(0, mouse_event->y());
      |                                                  ~~~~~~~~~~~~~~^~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/QMouseEvent:1,
                 from /home/alvin/pool/DSView/DSView/pv/view/view.cpp:28:
/usr/include/x86_64-linux-gnu/qt6/QtGui/qevent.h:220:16: note: declared here
  220 |     inline int y() const { return qRound(position().y()); }
      |                ^
In file included from /home/alvin/pool/DSView/DSView/pv/view/trace.h:34,
                 from /home/alvin/pool/DSView/DSView/pv/view/signal.h:38,
                 from /home/alvin/pool/DSView/DSView/pv/view/logicsignal.h:27,
                 from /home/alvin/pool/DSView/DSView/pv/view/logicsignal.cpp:25:
/home/alvin/pool/DSView/DSView/pv/view/selectableitem.h:39:31: error: invalid use of incomplete type ‘class QObject’
   39 | class SelectableItem : public QObject
      |                               ^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qpainter.h:11,
                 from /usr/include/x86_64-linux-gnu/qt6/QtGui/QPainter:1,
                 from /home/alvin/pool/DSView/DSView/pv/view/signal.h:29:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
In file included from /home/alvin/pool/DSView/DSView/pv/view/trace.h:35:
/home/alvin/pool/DSView/DSView/pv/view/dsldial.h:33:24: error: invalid use of incomplete type ‘class QObject’
   33 | class dslDial : public QObject
      |                        ^~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qscopedpointer.h:65:7: note: forward declaration of ‘class QObject’
   65 | class QObject;
      |       ^~~~~~~
[ 36%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/cursor.cpp.o
make[2]: *** [CMakeFiles/DSView.dir/build.make:1004: CMakeFiles/DSView.dir/DSView/pv/view/signal.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/DSView.dir/build.make:1032: CMakeFiles/DSView.dir/DSView/pv/view/logicsignal.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

```

priv, 2023-05-31:
After including QObject
```
[  0%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/data/mathstack.cpp.o
[  0%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dialogs/fftoptions.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/mathtrace.cpp.o
[  0%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/storesession.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/toolbars/titlebar.cpp.o
[  1%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/mainframe.cpp.o
[  2%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/widgets/border.cpp.o
[  2%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dialogs/dsmessagebox.cpp.o
/home/alvin/pool/DSView/DSView/pv/toolbars/titlebar.cpp: In member function ‘virtual void pv::toolbars::TitleBar::mousePressEvent(QMouseEvent*)’:
/home/alvin/pool/DSView/DSView/pv/toolbars/titlebar.cpp:202:41: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
  202 |             _clickPos = event->globalPos();
      |                         ~~~~~~~~~~~~~~~~^~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/QMouseEvent:1,
                 from /home/alvin/pool/DSView/DSView/pv/toolbars/titlebar.cpp:30:
/usr/include/x86_64-linux-gnu/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
/home/alvin/pool/DSView/DSView/pv/toolbars/titlebar.cpp: In member function ‘virtual void pv::toolbars::TitleBar::mouseMoveEvent(QMouseEvent*)’:
/home/alvin/pool/DSView/DSView/pv/toolbars/titlebar.cpp:214:47: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
  214 |        int x = _oldPos.x() + (event->globalPos().x() - _clickPos.x());
      |                               ~~~~~~~~~~~~~~~~^~
/usr/include/x86_64-linux-gnu/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
/home/alvin/pool/DSView/DSView/pv/toolbars/titlebar.cpp:215:47: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
  215 |        int y = _oldPos.y() + (event->globalPos().y() - _clickPos.y());
      |                               ~~~~~~~~~~~~~~~~^~
/usr/include/x86_64-linux-gnu/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
[  2%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dialogs/shadow.cpp.o
[  2%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dialogs/dsdialog.cpp.o
[  3%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dialogs/interval.cpp.o
[  3%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/prop/binding/probeoptions.cpp.o
[  3%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/viewstatus.cpp.o
[  4%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/dialogs/lissajousoptions.cpp.o
/home/alvin/pool/DSView/DSView/pv/storesession.cpp: In member function ‘bool pv::StoreSession::meta_gen(pv::data::Snapshot*, std::string&)’:
/home/alvin/pool/DSView/DSView/pv/storesession.cpp:462:24: warning: unused variable ‘sdi’ [-Wunused-variable]
  462 |     const sr_dev_inst *sdi = NULL;
      |                        ^~~
[  4%] Building CXX object CMakeFiles/DSView.dir/DSView/pv/view/lissajoustrace.cpp.o
/home/alvin/pool/DSView/DSView/pv/storesession.cpp: In member function ‘bool pv::StoreSession::json_decoders(QJsonArray&)’:
/home/alvin/pool/DSView/DSView/pv/storesession.cpp:1082:56: error: use of deleted function ‘QVariant::QVariant(T) [with T = char*; typename std::enable_if<disjunction_v<std::is_pointer<_Tp>, std::is_member_pointer<_Tp> >, bool>::type <anonymous> = false]’
 1082 |                 dec_obj["id"] = QJsonValue::fromVariant(d->id);
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/qlocale.h:7,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qcalendar.h:10,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qdatetime.h:11,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/QDateTime:1,
                 from /home/alvin/pool/DSView/DSView/pv/sigsession.h:32,
                 from /home/alvin/pool/DSView/DSView/pv/storesession.cpp:27:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:199:5: note: declared here
  199 |     QVariant(T) = delete;
      |     ^~~~~~~~
/home/alvin/pool/DSView/DSView/pv/storesession.cpp:1087:58: error: use of deleted function ‘QVariant::QVariant(T) [with T = char*; typename std::enable_if<disjunction_v<std::is_pointer<_Tp>, std::is_member_pointer<_Tp> >, bool>::type <anonymous> = false]’
 1087 |                 stack_obj["id"] = QJsonValue::fromVariant(d->id);
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:199:5: note: declared here
  199 |     QVariant(T) = delete;
      |     ^~~~~~~~
make[2]: *** [CMakeFiles/DSView.dir/build.make:1494: CMakeFiles/DSView.dir/DSView/pv/storesession.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/boost/bind.hpp:30,
                 from /home/alvin/pool/DSView/DSView/pv/prop/binding/probeoptions.cpp:23:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

```

dreamsource-tai, 2023-05-31:
@priv 
Don't install the qt6, install qt5 with pack qt5-default or qtbase5-dev.

priv, 2023-06-01:
Thanks, compilation passed with qt5

closed 2023-06-01 by priv