DreamSourceLab/DSView #636: DSCope U3P100 with DSLogic U3Pro16 connected

hondachuang, 2023-06-04:
1. When I connected both DSCope U3P100 and DSLogic U3Pro16 on my laptop, the DSCope was then disconnected.

2. If I started one DSView and connected it to DSCope, the connection was broken when I started another DSView. The connection cannot be recovered even the second DSView was closed. I had to reconnect the DSCope to get the connection back and everything captured was lost. 

What I want is to use TO of DSLogic as  trigger input of DSCope to see the relationship of the analog signal with the digital signals. Any suggestions?

Thank you very much.

Best Regards,
Honda

dreamsource-tai, 2023-06-05:
@hondachuang 
你好，我们v1.3正在测试中。这个版本解决了你的这个问题，到时候请留意。

hondachuang, 2023-06-05:
xx!!
如果能同r使用DSLogic 和 DScope就太棒了!

hondachuang, 2023-06-09:
> @hondachuang 你好，我们v1.3正在测试中。这个版本解决了你的这个问题，到时候请留意。

I can help for beta test to provide feedback. Thank you.

dreamsource-tai, 2023-07-03:
@hondachuang 
If your operating system is Linux, you can experience the latest features of DSView.