DreamSourceLab/DSView #644: Zoom issue between live capture and .dsl file

Mathias-Buder-Bose, 2023-06-14:
Today I noticed that there is a difference in min. allowed zoom level between a capture that just habe been taken and a capture that has been saved to a .dsl file and opened at a later point in time as shown below. The zoom level of the .dsl file should also be available in the live capture as it allows an greater overview of the data.

Live Capture
[image]

.dsl file
[image]

dreamsource-tai, 2023-06-15:
@Mathias-Buder-Bose 
Can you upload the data file? The dsl file.

Mathias-Buder-Bose, 2023-06-15:
@dreamsource-tai sure, here you go: [DSLogic-la-230614-131502.zip](https://github.com/DreamSourceLab/DSView/files/11756195/DSLogic-la-230614-131502.zip)

Mathias-Buder-Bose, 2023-06-15:
Just noticed that today, the zoom level with live capture seams to be fine now. The only thing I did was disconnect the analyzer from the USB and reconnect it.

[image]