DreamSourceLab/DSView #650: Bug report - Interval time calculation error

citizen6874, 2023-06-19:
The time calculated by the blue floating tooltip is out by a factor of 10.

The tooltip gives Width, Period, Frequency and Duty Cycle.  The time is 10x too long for the interval, and subsequent dependant calculations are thus also wrong.

See attached screen grab.

[image]

Also, what happened to the curved line allowing measurements between events in different channels? I can't seem to activate that now.

Cheers,
Citizen6874

dreamsource-tai, 2023-06-19:
@citizen6874 
Can you upload youre test data?

citizen6874, 2023-06-19:
Sorry but I didn’t keep it.

citizen6874, 2023-06-19:
After restarting the software the bug was gone, and the curved brown measuring lines were working again.